thonny/thonny #1542: GenericMeta error message on running program

timsoftgit, 2021-01-14:
when running thonny 3.3.2 from a terminal, when I run a simple print("hello world") program I see the following error message on the terminal.
`WARNING thonny.plugins.mypy: MyPy: Traceback (most recent call last):
  File "/usr/lib64/python3.7/runpy.py", line 193, in _run_module_as_main
     "__main__", mod_spec)
  File "/usr/lib64/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/lib64/python3.7/site-packages/mypy/__main__.py", line 4, in <module>
    from mypy.main import main
  File "/usr/lib64/python3.7/site-packages/mypy/main.py", line 11, in <module>
    from typing_extensions import Final, NoReturn
  File "/usr/lib64/python3.7/site-packages/typing_extensions.py", line 3, in <module>
    from typing import (  # noqa
ImportError: cannont import name 'GenericMeta' from 'typing' (/usr/lib64/python3.7/typing.py)`
I didn't get this with v3.3.1 any ideas. I am using system libraries.

aivarannamaa, 2021-01-14:
MyPy is used by Thonny for checking user code for typing errors.

In your case it looks like the version of MyPy and its optional extra `typing_extensions`. AFAIK it is not useful in Python 3.7, so you could try removing it (`sudo apt remove python3-typing-extensions` if you have a Debian based system).

timsoftgit, 2021-01-14:
I tried removing mypy_extensions and mypy but that gave more errors, so I restored them. I then tried adding typing and typing_extensions but that didn't change the situation. I am using the same mypy (0.790) and mypy_extensions (0.4.3) as I was using with 3.3.1. typed_ast was recently updated to 1.4.2 but that is the only change I can think of which may be significant apart from thonny itself.

aivarannamaa, 2021-01-16:
How did you install Thonny? At the moment I don't know how to reproduce this.

timsoftgit, 2021-01-18:
using the thonny slackbuild. see 
https://slackbuilds.org/slackbuilds/14.2/python/thonny/thonny.SlackBuild
(updated to the current version). As I provided the script, I can change it if needed; but it is there so that it is managed by slackware package management tools. Basically the dependencies (as listed in https://slackbuilds.org/slackbuilds/14.2/python/thonny/thonny.info), (which will have their own dependencies), satisfy the deps, and the thonny module is just copied over, and run with a simple script https://slackbuilds.org/slackbuilds/14.2/python/thonny/thonny.sh

aivarannamaa, 2021-01-18:
Sorry, I'm out of ideas. The problem seems to be outside of Thonny's code and its [requirements.txt](https://github.com/thonny/thonny/blob/master/requirements.txt). I haven't touched MyPy-related stuff in Thonny 3.3.

Does the problem go away when you revert to Thonny 3.3.1?

timsoftgit, 2021-01-21:
sorry for the delay. I tried going back to thonny 3.3.1 but am getting the same issue there now, which I am pretty sure I didn't get before (I always test my helloworld.py in thonny before pushing the updated slackbuild). I've removed typing_extensions  (it appears that it is not supposed to be needed for python3.7) but mypy calls typing-extensions from main.py, build.py, nodes.py and util.py and unfortunately if you just comment out the typing-extensions import lines mypy doesn't work. I did try rolling mypy back from 0.790 to 0.782 , then 0.781 and 0.780 but am still getting an error from mypy which is trying to import from typing_extensions. I might have to go through the deps again. What are the current deps for thonny?.

timsoftgit, 2021-01-21:
looking at typing.py there is a GenericMeta class, so I'm not sure why it can't be imported by typing_extensions.py

aivarannamaa, 2021-01-21:
Have you tried using `mypy` without Thonny?

timsoftgit, 2021-01-27:
I'm not sure how to do that. any pointers? I've tried
`mypy helloworld.py
Traceback (most recent call last):
  File "/usr/bin/mypy", line 11, in <module>
    load_entry_point('mypy==0.800', 'console_scripts', 'mypy')()
  File "/usr/lib64/python3.7/site-packages/pkg_resources/__init__.py", line 487, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib64/python3.7/site-packages/pkg_resources/__init__.py", line 2728, in load_entry_point
    return ep.load()
  File "/usr/lib64/python3.7/site-packages/pkg_resources/__init__.py", line 2346, in load
    return self.resolve()
  File "/usr/lib64/python3.7/site-packages/pkg_resources/__init__.py", line 2352, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib64/python3.7/site-packages/mypy/__main__.py", line 6, in <module>
    from mypy.main import main
  File "/usr/lib64/python3.7/site-packages/mypy/main.py", line 11, in <module>
    from typing_extensions import Final, NoReturn
  File "/usr/lib64/python3.7/site-packages/typing_extensions.py", line 3, in <module>
    from typing import (  # noqa
ImportError: cannot import name 'GenericMeta' from 'typing' (/usr/lib64/python3.7/typing.py)`

Which seems to point to mypy or another module it uses. I'm not sure why GenericMeta is being called as python version is above 3.5 so it has been retired.
My target is getting thonny running without any command-line errors, I'll have a look at the issue trackers for the sub modules/projects but I don't know what has changed that causes the problem. typed_ast is the only thing that has changed recently (apart from mypy itself going to 0.800)

aivarannamaa, 2021-01-27:
Sorry, I'm not so familiar with MyPy's internals. You might get an answer from https://github.com/python/mypy/issues

timsoftgit, 2021-01-27:
I have looked further, and I have found an issue with the slackware packaging of typing_extensions, where it appears to be installing the python2 version of typing_extensions in both the python2 and the python3 location. Manually putting the right version in the right place fixes the problem, so I'll report to the packager and get it fixed. Sorry for the noise.

closed 2021-01-27 by timsoftgit