thonny/thonny #1549: Thonny fails to open the File>Open... dialog

ghcreed, 2021-01-17:
Thonny version 3.3.1 on Linux Mint 17 (32-bit MATE) fails to open the File>Open... dialog (Ctrl+O).
The directory browser that is used to select an alternative interpreter fails too.
Here is the output from the terminal where I launched Thonny (./thonny):
----
INFO    thonny: Thonny version: 3.3.1
This option is not available. Please see --help for all possible usages.
This option is not available. Please see --help for all possible usages.
----

aivarannamaa, 2021-01-17:
You seem to have problems with `zenity` program, which is used by Thonny for presenting file dialogs. Please send me the output of following two commands:

* `zenity --version`
* `zenity --help`

See #1550 for a work-around.

ghcreed, 2021-01-23:
The problem is solved by issue #1550's work-around.
I've attached an output file, ZenityInfo.txt.
[ZenityInfo.txt](https://github.com/thonny/thonny/files/5861179/ZenityInfo.txt)
Nice IDE, written with Tkinter, very impressive. Thank you.

closed 2021-01-23 by ghcreed