thonny/thonny #1550: Fedora Jam - Thonny runs only as root without errors  ENV MYPYPATH

Logger-Boy, 2021-01-17:
I've been playing with this for some time as a Python beginner. I can run Thonny from a terminal as root and it works. If I run it from the terminal as a user  it will not save or open files.  Interestingly enough it will ask to replace a file.  My reading suggests that an environment variable "MYPYPATH"  should be changed in many cases because including the directory "site-packages" can cause problems.  So far I have not found the variable or a way to change it, if indeed it is the problem. Here is the traceback message...
`Traceback (most recent call last):
  File "/usr/lib64/python3.9/tkinter/__init__.py", line 1884, in __call__
    return self.func(*args)
  File "/usr/lib/python3.9/site-packages/thonny/running.py", line 382, in cmd_run_current_script
    self.execute_current("Run")
  File "/usr/lib/python3.9/site-packages/thonny/running.py", line 337, in execute_current
    filename = get_saved_current_script_filename()
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 957, in get_saved_current_script_filename
    filename = editor.get_filename(force)
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 72, in get_filename
    self.save_file()
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 234, in save_file
    result = self.write_local_file(save_filename, content_bytes, save_copy)
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 247, in write_local_file
    f = open(save_filename, mode="wb")
FileNotFoundError: [Errno 2] No such file or directory: 'WINDOW DECORATIONS RELOADED\n/home/kevin/Code/Python/Kevin.py'

`
help would be greatly appreciated.  Namaste, Logger-Boy

aivarannamaa, 2021-01-17:
It looks like you happen to have buggy `zenity` program (https://bugs.kde.org/show_bug.cgi?id=431365), which Thonny uses for presenting file dialogs.

The work-around (until your zenity gets fixed) is to switch to Tk file dialogs (Tools => Options => General => Use Tk file dialogs).

cybermito, 2021-01-21:
> It looks like you happen to have buggy `zenity` program or a zenity's dependency (https://bugs.kde.org/show_bug.cgi?id=431365), which Thonny uses for presenting file dialogs.
> 
> The work-around (until your zenity gets fixed) is to switch to Tk file dialogs (Tools => Options => General => Use Tk file dialogs).

Good morning, the option to change to tk, where can I find it? I've looked at Thonny but she's nowhere to be found, or I can't find her.
Thank you.

aivarannamaa, 2021-01-21:
Oh, right, Fedora doesn't offer newest Thonny yet. In this case you need to wait until `zenity` starts behaving properly (did you try updating your system?) or `dnf unistall thonny` and `bash <(wget -O - https://thonny.org/installer-for-linux)` instead.

cybermito, 2021-01-22:
Sorry, at first I did not understand the message correctly and I reinstalled the latest version of Thonny, 3.3.3 and indeed in this version the option to change to TK comes out, so I did it and it works.
Thank you very much, the problem you had is resolved.
Regards.

aivarannamaa, 2021-01-22:
Great! I'm closing this then, as it's KDE's issue, not Thonny's

closed 2021-01-22 by aivarannamaa