thonny/thonny #1554: Cannot start multiple Thonny instances on Mac

pythonprocourse, 2021-01-19:
Even though the "Allow only one instance of Thonny" option is unchecked in Thonny, another instance of Thonny cannot be started on Mac. Same behaviour in two Mac computers:

First computer: Thony version: 3.3.1, Mac OSX version 10.15.7
Second computer: Thonny version 3.3.2, Mac OSX version 11.1

aivarannamaa, 2021-01-19:
Thank you for the report!

I could reproduce this problem. Probably related to #1555

At the moment I don't know how to solve this. The work-around is to launch second instance from command line, eg `/Applications/Thonny.app/Contents/MacOS/thonny`.

ux-lic, 2024-07-04:
Worth noting that this work around only works if you launch both apps from cmd line. i.e if you already have an icon launched version running close it first.

`open -n -a thonny.app`

JetForMe, 2024-08-31:
Thonny really ought to just support multiple target devices in one instance of the app.