thonny/thonny #1557: "Wrong type of command" error when submitting user input

aivarannamaa, 2021-01-20:
https://raspberrypi.stackexchange.com/questions/120214/input-called-from-gpio-input-button-press-returns-error

aivarannamaa, 2021-01-20:
I can reproduce when the main thread of following code completes, Thonny gives me command prompt and I'm entering expected input there:

```
import time
import threading

def work():
    print(input("> "))

threading.Thread(target=work).start()
```

AgribotGit, 2022-12-20:
I have the same defect. This is a serious defect that will impact a lot of users. You should prioritise a solution.

aivarannamaa, 2022-12-28:
> I have the same defect.

@AgribotGit, do you mean you get the same error with program using threads?

agribot2, 2023-09-06:
I have experienced this defect when trying to use python threads in Thonny. Please help to fix the defect.