thonny/thonny #1565: Slow problems starting the M5Stack CORE2 program.

fraramon, 2021-01-22:
Hi, I am having a lot of slowness issues starting programs in M5Stack CORE2.

The problem is the following, every time I run a program in thonny it reboots the device, and the problem is that the CORE2 reset is a bit slow.

I have done tests with MU editor and it goes fast because when running a program it does not restart the device.

fraramon, 2021-01-22:
I have tested version 3.2.7 and it works fine, that is, it is from the new version.

fraramon, 2021-01-22:
I have tried it in the latest version and it has the problem.

I think that as a conclusion is that the device must be reset as little as possible, because the restart is very very slow, in any case simulate a Ctrl + C that is more effective to cut the main loop and nothing else.

aivarannamaa, 2021-01-22:
Thank you for the report!

> I have tested version 3.2.7 and it works fine, that is, it is from the new version.

This is expected, as soft-reboot before run was added in Thonny 3.3. 

Unfortunately I don't have a CORE2 M5Stack myself. You were saying "reboots the device" -- are you sure the device gets reset or just the MicroPython virtual machine (ie soft-reboot)? Actually, I'm not sure whether it matters, because if there is lot to do in `boot.py`, then even soft-reboot can be slow.

I guess I could make soft-rebooting before Run optional. (Can't just remove it because some people have explicitly asked for it.)

fraramon, 2021-01-22:
Hello aivarannamaa, 
the problem is with the soft restart or the other.
The only thing that works well is Ctrl + C.
Fran,

fraramon, 2021-01-22:
If you want me to do any test you just have to tell me.

fraramon, 2021-01-22:
Hello aivarannamaa,

I have made you a summary of behaviors in case it helps you:

Version 3.2.7
Ctrl-D: Restart micropython runtime and my infinite loop (14 seconds)
Ctrl-F2: Restart and black screen with cursor REPL (6 seconds).
Ctrl-C: Msg KeyboardInterrupt and exit of my infinite loop.
F5: Run the program, but if I don't press "Ctrl-C" before, the process is locked, it run very quickly (it would be nice if it did it automatically)

Version 3.3.3
Ctrl-D: Restart micropython runtime and my infinite loop (14 seconds)
Ctrl-F2: Restart and my infinite loop (12 seconds).
Ctrl-C: I have to press twice to msg keyboard interrupt and exit of my infinite loop.
F5: Run the program, but it restart first with long time (12 seconds).

La version 3.2.7 se esta comportando mejor, es mas rápida de ejecutar, solo le faltaría un Ctrl-C automático antes de ejecutar F5.

If I can prove anything, feel free to tell me.

Fran,

eugeniobaglieri, 2021-03-24:
I'm having same issue, could you tell with in which commit you enabled soft-reset by default?
So we can manually edit the code to disable it.
Thank you

aivarannamaa, 2021-03-24:
@eugeniobaglieri, you could try removing this line: https://github.com/thonny/thonny/blob/37c58e4bd9d1dc7455b641e2bdc2b7bc4216bb86/thonny/plugins/micropython/bare_metal_backend.py#L954

eugeniobaglieri, 2021-03-24:
Done, and it works! 
For who has a Mac, this is the file path:

/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py

aivarannamaa, 2021-12-01:
Thonny 4.0 will have the option to disable soft-reboot before run.

TODO: note to myself - use `%run` instead of `%Run` if this option is selected.

closed 2022-05-14 by aivarannamaa