thonny/thonny #1566: TypeError: unsupported operand type(s) for +: 'method' and 'str'

klint-k, 2021-01-22:
Traceback (most recent call last):
  File "/usr/lib/python3.9/tkinter/__init__.py", line 1884, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pip_gui.py", line 655, in _handle_install_file_click
    filename = askopenfilename(
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1920, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1960, in askopenfilename
    args = cls._convert_common_options("Open file", **options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1994, in _convert_common_options
    filename = _options_to_zenity_filename(options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2036, in _options_to_zenity_filename
    return options["initialdir"] + os.path.sep
TypeError: unsupported operand type(s) for +: 'method' and 'str'

was not fixed

aivarannamaa, 2021-01-23:
Which Thonny version are you using? It should be fixed in 3.3.3.

aivarannamaa, 2021-01-23:
Also, note that installing Pico plug-ins is not required when you are using Thonny 3.3.3 or later. Furthermore, the plug-in is not compatible with Thonny 3.2 and older.

aivarannamaa, 2021-01-30:
Duplicate of #1583

closed 2021-01-30 by aivarannamaa