thonny/thonny #1570: feature request: A filemanager for micropython devices

hubertmeier, 2021-01-23:
It would be nice to have a **file manager** which can copy, rename, delete files on the micropython filesystem. Saving is perfect, for deleting and listing we have to use the command line ...

Thank you

aivarannamaa, 2021-01-23:
Have you tried "View => Files"?

hubertmeier, 2021-01-24:
Hi,

yes, i tried "files", i can do the same operations like delete oder open like on the command line, but not copy (from PC) or rename the file on the device or missed i something? Perhaps this has to be performed in several steps:

rename: download the file to thonny, delete it on device, store it on device with the new name
copy: open it in thonny, store it on the device with the same name as on PC

thank you

aivarannamaa, 2021-01-24:
Here is an example of how to copy a file from PC to the device: 
[image]

Rename can't be done in the file browser yet, but if you double-click on a remote (text) file and open it in Thonny's editor, then you can use File => Move / rename.

hubertmeier, 2021-01-25:
OK, thank you very much for your patience, i did not expected "upload to /" at the file properties ...

thank you very much for your patience

aivarannamaa, 2021-01-27:
I guess I should make these commands more discoverable. I'll think about this.