thonny/thonny #1571: Unable to bind additional callbacks to <BackSpace> key

Samigirma1, 2021-01-24:
I was developing a Thonny plug-in for a class project. I need to add an additional callback to all **\<KeyPress\>** or **\<Key\>** events for the current editor. (code below) But the **\<BackSapce\>** key is not triggering the callback on Mac, Linux (Ubuntu), and Windows. I also tried binding the callback specifically for the key. That didn't work either.

```
#...
text_widget = get_workbench().get_editor_notebook().get_current_editor().get_text_widget()
id = text_widget.bind("<KeyPress>", self.bounce, True) # self.bounce defined else where 
#...
```

Note: I tried the callback on a simple Tkinter window with a text widget and it worked just fine.

aivarannamaa, 2021-01-24:
Thonny has assigned [special code](https://github.com/thonny/thonny/blob/6ebabec72149bcd25fbcdf7db0c226b26a2b4c0f/thonny/tktextext.py#L239) for Backspace key and in order to cancel the default action, it returns `"break"` from the event handler.

You could try passing `False` as 3rd argument to `bind`, so that previous handlers get removed, but in this case you lose the smart backspace feature (unless you finally call the original handler yourself in the end of your handler).

Alternatively, you could patch the `perform_smart_backspace` method in `SyntaxText`. Here is an example of this approach: https://github.com/thonny/thonny/blob/master/thonny/plugins/autocomplete.py#L343

What is your Backspace handler supposed to do?

Samigirma1, 2021-01-24:
It's supposed to send over network the range of indices deleted and the position of the insertion cursor after the delete.

aivarannamaa, 2021-01-24:
I suggest you look at `thonny.plugins.event_logging`. Everything happening in the editors are captured by this module. Pressing Backspace ends up as `TextDelete` event. See https://github.com/thonny/thonny/wiki/User-action-logs for the background.

Samigirma1, 2021-01-24:
Awesome! Thank you so much. Is there documentation for Thonny's virtual events and the events' data attributes?

aivarannamaa, 2021-01-24:
Unfortunately they are not documented, but you can switch on logging and inspect the resulting logs (or use the replayer). See the wiki page for instructions.

closed 2021-02-12 by Samigirma1

aivarannamaa, 2023-02-22:
@Samigirma1, it seems that you have completed your project successfully. Do you happen to have a write-up explaining your approach and difficulties you met?

Samigirma1, 2023-03-01:
It's been 2 years. So, I don't remember the details of what we did. However, quickly glossing through our repo, I think we simply patched the insert and delete methods of the `EnhancedText` class with a function that emits an event with the arguments. We had a listener inspect the arguments and take the appropriate action