thonny/thonny #1573: cannot import name 'Serial' from 'serial'

Nicobibi, 2021-01-24:
I got a Rasberry Pi 3B+ up to date with python libraries 2.7 and 3.7 also up to date and I run a python code with Thonny 3.3.2. I try to run that code :

`from serial import Serial
import RPI.GPIO as GPIO
import os, time
from gsmHat import GSMHat, SMS, GPS`

But I got this issue :

> ImportError: cannot import name 'Serial' from 'serial' (/home/pi/.local/lib/python3.7/site-packages/serial/init.py)

So I tried to uninstall and install pyserial on python 2.7 and 3.7

`pip uninstall pyserial
pip install pyserial
pip3 uninstall pyserial
pip3 install pyserial`

It didn't work so I tried with super user command :

`sudo pip uninstall pyserial
sudo pip install pyserial
sudo pip3 uninstall pyserial
sudo pip3 install pyserial`

I also try to rename my *.py code file to avoid a name overlap. Then I tied to remove "from serial import Serial" from the code :

`import RPI.GPIO as GPIO
import os, time
from gsmHat import GSMHat, SMS, GPS`

and I had the same issue that I had on serial module on RPI.GPIO one :

> ModuleNotFoundError: No module named 'RPI'

So as mosul issue came up with two modules I tried to reinstall python :

`sudo apt-get install python-pip`

But python was already up to date as I saw it when I tiped :

`pip freeze | grep Rpi pip3 freeze | grep Rpi`

Who return something in both python 2.7 and 3.7 :

> GPi.GPIO==0.7.0

so I looked in **Thonny > Tools > Manage Packages > Serial** to check that Thonny sees libraries in my /home/usr/python3 directory and Thonny sees botn serial and RPi.GPIO libraries.

So it seems that all libraries are install and compilation app sees them but do not use them.

Any idea please?

Thanks a lot.

aivarannamaa, 2021-01-24:
> ImportError: cannot import name 'Serial' from 'serial' (/home/pi/.local/lib/python3.7/site-packages/serial/init.py)

From this I see that you have (or had) something called `serial` in your **user site-packages** directory.

> so I looked in Thonny > Tools > Manage Packages > Serial

Note that the *distribution* for serial communication is named `pyserial`, not `serial` (unlike its *package* name, which is indeed `serial`). The *distribution* named `serial` is something else: https://pypi.org/project/serial/

Nicobibi, 2021-01-24:
Hello aivarannamaa, 

You're right, I didn't mention that I also checked pyserial package 
> honny > Tools > Manage Packages > pySerial
I also tryed to replace 
`from serial import Serial`
by
`from pyserial import Serial`
The issue is still the same...

aivarannamaa, 2021-01-24:
Nope, that's not what I meant. The correct package name is `serial`, so `from serial import Serial` is proper form.

I just suspect that the **distribution** named `serial` is not what you want, because it may have hidden up the **package** `serial` from distribuition named `pyserial`. So, unless you are sure you need "A framework for serializing/deserializing JSON/YAML/XML into python class instances and vice versa", then go to "Tools => Options => Manage packages", select `serial` from the left and uninstall it. Most likely this will fix your problems.

Nicobibi, 2021-01-24:
Good idear! 
I unstalled serial lib and kept pyserial one and I still have the same error message : 

> NameError: name 'serial' is not define

[image]

[image]
[image]

aivarannamaa, 2021-01-24:
Try following program:

```
import serial
print(serial.__file__)
```

What does it print?

Nicobibi, 2021-01-24:
That print : 

> /usr/lib/python3/dist-packages/serial/__init__.py

aivarannamaa, 2021-01-24:
Could you post the contents of `/usr/lib/python3/dist-packages/serial/__init__.py` here?

Nicobibi, 2021-01-25:
Here is ~/serial/__init__.py content :

```
#!/usr/bin/env python
#
# This is a wrapper module for different platform implementations
#
# This file is part of pySerial. https://github.com/pyserial/pyserial
# (C) 2001-2017 Chris Liechti <cliechti@gmx.net>
#
# SPDX-License-Identifier:    BSD-3-Clause

import sys
import importlib

from serial.serialutil import *
#~ SerialBase, SerialException, to_bytes, iterbytes

__version__ = '3.4'

VERSION = __version__

# pylint: disable=wrong-import-position
if sys.platform == 'cli':
    from serial.serialcli import Serial
else:
    import os
    # chose an implementation, depending on os
    if os.name == 'nt':  # sys.platform == 'win32':
        from serial.serialwin32 import Serial
    elif os.name == 'posix':
        from serial.serialposix import Serial, PosixPollSerial, VTIMESerial  # noqa
    elif os.name == 'java':
        from serial.serialjava import Serial
    else:
        raise ImportError("Sorry: no implementation for your platform ('{}') available".format(os.name))

protocol_handler_packages = [
    'serial.urlhandler',
]

def serial_for_url(url, *args, **kwargs):
    """\
    Get an instance of the Serial class, depending on port/url. The port is not
    opened when the keyword parameter 'do_not_open' is true, by default it
    is. All other parameters are directly passed to the __init__ method when
    the port is instantiated.

    The list of package names that is searched for protocol handlers is kept in
    ``protocol_handler_packages``.

    e.g. we want to support a URL ``foobar://``. A module
    ``my_handlers.protocol_foobar`` is provided by the user. Then
    ``protocol_handler_packages.append("my_handlers")`` would extend the search
    path so that ``serial_for_url("foobar://"))`` would work.
    """
    # check and remove extra parameter to not confuse the Serial class
    do_open = not kwargs.pop('do_not_open', False)
    # the default is to use the native implementation
    klass = Serial
    try:
        url_lowercase = url.lower()
    except AttributeError:
        # it's not a string, use default
        pass
    else:
        # if it is an URL, try to import the handler module from the list of possible packages
        if '://' in url_lowercase:
            protocol = url_lowercase.split('://', 1)[0]
            module_name = '.protocol_{}'.format(protocol)
            for package_name in protocol_handler_packages:
                try:
                    importlib.import_module(package_name)
                    handler_module = importlib.import_module(module_name, package_name)
                except ImportError:
                    continue
                else:
                    if hasattr(handler_module, 'serial_class_for_url'):
                        url, klass = handler_module.serial_class_for_url(url)
                    else:
                        klass = handler_module.Serial
                    break
            else:
                raise ValueError('invalid URL, protocol {!r} not known'.format(protocol))
    # instantiate and open when desired
    instance = klass(None, *args, **kwargs)
    instance.port = url
    if do_open:
        instance.open()
    return instance

```

aivarannamaa, 2021-01-25:
If you really posted `~/serial/__init__.py`, then this doesn't help us, because according to `serial.__file__` the file that gets used is `/usr/lib/python3/dist-packages/serial/__init__.py`

Nicobibi, 2021-01-25:
It's the one : I posted /usr/lib/python3/dist-packages/serial/__init__.py 
Sorry for missunderstanding.

aivarannamaa, 2021-01-27:
In this case I don't know what is wrong. Does your program run fine outside of Thonny?

Nicobibi, 2021-01-27:
I just tried with Linux control panel and I have the same issue.

aivarannamaa, 2021-01-27:
Linux control panel? Did you mean Terminal? Command line? 

If it's not Thonny specific problem, then I can't help you further. I suggest you ask at Raspberry Pi forum.

Nicobibi, 2021-01-27:
OK... Now it works with both Linux command and Thonny. The only thing I did is installing PySide2 package for Qtcreator yesturday but it's got nothing to deal with serial lib... I am puzzled... 
Thanks for trying to help. It seems I was just missing a bit of magic...

aivarannamaa, 2021-01-27:
I'm glad you found the required magic :)

closed 2021-01-27 by aivarannamaa

normal7070, 2022-10-21:
i have the same issue any idea?