thonny/thonny #1574: strange behaviour of control+shift+right_arrow on the last word of a line

omfgwhydoineedausername, 2021-01-26:
for example, given the python code:

print(cam.some_data)
                ^
with the cursor here
    
the editor will jump the cursor to the beginning of the first token on the next line, when it would feel more natural to simply select the word 'some_data'.

I'm 'feels' unusual to me after using visual studio for many years.

[image]

aivarannamaa, 2021-01-27:
Thank you for the report!

I can't reproduce this problem with Thonny 3.3.3 in Ubuntu. Which OS and which Thonny version are you using?

omfgwhydoineedausername, 2021-01-28:
Hi Aivar
I'm using Thonny 3.3.3 on Windows 10 (64-bit) and Python 3.9.1 (64 bit). 

I can see ctrl+shift+right arrow is moving to the start of the next token but when selecting items with ctrl+shift+right arrow it would be much more intuitive if it just selected/deselected the current token.
Hopefully you can allow us to change this or customize it as I really prefer and enjoy Thonny over visual studio code.
Thanks!

aivarannamaa, 2021-01-28:
I could confirm this in Windows.

Currently the implementation for this comes from Thonny's user interface toolkit (Tk). Before doing anything myself, I'll wait until February, when a new Python is supposed to be released together with newer Tk.

omfgwhydoineedausername, 2021-01-29:
Thanks for looking into it.