thonny/thonny #1577: error Backend terminated or disconnected. Use 'Stop/Restart' to restart.

pythonbeginner17, 2021-01-28:
I am pretty new to this thonny and python thing. So I tried to run cv2 on my Thonny 3.3.2 on mac os big sur but the error backend terminated keeps appears even though I restart the app a few times.

aivarannamaa, 2021-01-28:
Thank you for the report!

Does this happen with all programs or only when using cv2? Can you post a minimal program that causes this problem?

closed 2021-02-17 by aivarannamaa

amogorkon, 2021-02-18:
Hi,

I'm having the same problem. However, I'm not using cv2. 

Thonny 3.3.3 running Python 3.9.0.

It is very hard to diagnose this issue because it crashes while stepping into a constructor of a pure Python class, with no means of getting more info.

aivarannamaa, 2021-02-18:
@amogorkon, thank you for this info! When you find the minimal program, then please post it under a new issue!

binghe2402, 2021-04-14:
I have the same problem. I run cv2 (4.5.1.48) on my Thonny 3.3.6 on macOS 11.2.3 (20D91) , and when I try to use the camera the error occurred(Backend terminated or disconnected. Use 'Stop/Restart' to restart.)
My code is 
```python
import cv2
cap = cv2.VideoCapture(0)
```

if I do not use camera, it works.
```python
import cv2
cap = cv2.VideoCapture("test_video.mp4")
while(True):
    ret , frame = cap.read()
    cv2.imshow('frame',frame)
    if cv2.waitKey(1) &0xFF ==ord('q'):  
    	break
cap.release()
cv2.destroyAllWindows()
```