thonny/thonny #1579: More options for recommended max line length margin line

FidgetyNinja, 2021-01-28:
Would it be possible to add the max line length margin line into the theme options? I like using the Tomorrow Night Bright syntax theme, but the margin line is nearly impossible to see on my monitor. 

I would like to:
- Change color, thickness, or brightness of the margin line (one or more of those things to make it more visible). 
- Enter custom character limits like 72 since that's recommended in PEP 8.

aivarannamaa, 2021-01-29:
Thank you for the report!

> I like using the Tomorrow Night Bright syntax theme, but the margin line is nearly impossible to see on my monitor.

I'll make sure the line gets suitable color for each of the themes.

> Change color, thickness, or brightness of the margin line (one or more of those things to make it more visible).

I prefer not to add too many configuration items to the configuration dialog. But I think I should make the line thicker automatically in case of high DPI display. Also, I don't mind adding a hidden configuration variable, which can be changed only by editing the configuration.ini or by a new "Advanced options" dialogs in the upcoming Thonny 4.0

FidgetyNinja, 2021-01-29:
That sounds great, thanks! Also, do you have plans to update the distro packages? I'm using Thonny on a Pi 4 with the standard raspberry pi os

aivarannamaa, 2021-01-29:
Currently Raspberry has latest Thonny and usually they will update it quite soon after I release a new version.