thonny/thonny #1581: Cannot Stop/Reset if main.py is saved in ESP32 on thonny 3.3.3

amirbature, 2021-01-28:
I noticed that when ever I saved a file with main.py in device, it does'nt stop/reset or even disconnect. I must reload the firmware file if I want to edit the code.

aivarannamaa, 2021-01-29:
Thank you for the report!

What does your main.py does?

I've heard of inability to interrupt the code if it's certain type of sleep is used (#1313), but I'm really surprised that you can't disconnect. Do you get an error, or simply nothing happens when you select Run => Disconnect?

lazyoldjack, 2021-01-29:
I couldn't get 3.3.3 to work with my ESP32 reliably at all so have gone back to 3.3.1. Pico will have to wait!

amirbature, 2021-01-29:
> Thank you for the report!
> 
> What does your main.py do?
> 
> I've heard of inability to interrupt the code if it's certain type of sleep is used (#1313), but I'm really surprised that you can't disconnect. Do you get an error, or simply nothing happens when you select Run => Disconnect?

The program will just pause for 2 seconds and it will just continue running the main.py program.

lazyoldjack, 2021-01-29:
Well, my main.py it sits in a loop scanning touch pins, but there is 
also a sleep(0.1) in that loop but I am not sleeping any of the 
processors. Hitting the stop button in Thonny just brought up "break" 
lines in the Shell window but didn't bring up a prompt or any text. And 
yes, my program kept running.

I'm afraid I cannot try anything else now as I have reverted to 3.3.1 so 
I can continue what I was working on with ESP32 before my Pico arrived 
in the post. I added my comments so that you knew that someone else had 
also experienced problems and that Amir was not a one-off :o)

I will follow with interest as I am keen to keep my Thonny up to date; 
it's a great piece of software.

------------------------------------------------------------------------

aivarannamaa, 2021-01-31:
@lazyoldjack, in Thonny 3.3.1 the communication was based on MicroPython paste-mode REPL, in 3.3.2 it was reverted to raw mode. It looks like different devices prefer different communication modes. Here is how you can switch ESP32 to `paste` submit mode in Thonny 3.3.3: https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration

aivarannamaa, 2021-01-31:
@amirbature, do you get an error, or simply nothing happens when you select Run => Disconnect?

amirbature, 2021-01-31:
Nothin happens, it just continue executing the code in main.py

lazyoldjack, 2021-01-31:
> @lazyoldjack, in Thonny 3.3.1 the communication was based on MicroPython paste-mode REPL, in 3.3.2 it was reverted to raw mode. It looks like different devices prefer different communication modes. [...]

@aivarannamaa OK,reinstalled 3.3.3... 
I tried the different submit modes but tbh I couldn't see much difference. Pressing the STOP button just brings up the break lines **and restarts the program**, but using _**Run->Disconnect THEN pressing STOP works**_. Although, as a newcomer to Thonny, I have always used the STOP button in the past, I have also discovered that Ctrl-C **will** stop the program (obviously I guess!) and so I can use 3.3.3 OK. I was thrown by the STOP button no longer working :o)

I hope some of this is helpful? Note to self: never become overdependent on the mouse; remember the kbd commands!

aivarannamaa, 2021-02-01:
@lazyoldjack, thank you for this information. It is useful indeed.

The "stop" button is meant to be "stronger" than Ctrl+C (for example, it sends several Ctrl-Cs to the device among other things). I need to investigate what's wrong with it

r0the, 2021-02-11:
I had the same issue with ESP32 boards. With the combination MicroPython < 1.14 and Thonny 3.3 I always got "Device is busy or does not respond..." in the shell. I also wasn't able to flash a new firmware.

Thanks to the hint to the config settings, I set `submit_mode = raw` and was able to connect to the device again, but only by resetting the device with the reset button. MicroPython 1.14 seems to work fine with Thonny 3.3.

@aivarannamaa Maybe you could add a hint to the "Device is busy" message so that people with older MicroPython versions understand what is going on. Even better but probably not so easy would be to automatically change the comm mode to "raw" when the device is not responding.

jjcfrank, 2021-05-26:
This is not a great solution but definitely works until a proper fix. I tried the above (submit_mode = raw) but did not work btw. I installed uPyCraft as I had no other option in terms of IDE. Bear in mind I had Thonny in the background so I could see was was not responding. To my surprise, as soon as I pressed "connect to Esp32" in uPyCraft then magically Thonny was working again. Weird. But it works.

lutfiblog, 2021-06-20:
well i have same problem with uploading to main.py, i try interupt execution CTRL+C its work and make it editable

TKrom, 2021-09-26:
Dear authors,

I am experiencing exactly the same problems as depicted in this post. In short, when I use Thonny for running my simple test code (I am a beginner), the program runs fine; only when I want to stop the code from running by clicking the stop button in Thonny, nothing happens. Luckily ctrl + c does work, albeit for some situations. I am using the newest version of Thonny (3.3.13 for Windows), I have an ESP32 devkit V1 from Doit, and experience problems with all last stable versions of MicroPython (esp32-20210902-v1.17.bin, esp32-20210623-v1.16.bin, esp32-20210418-v1.15.bin, and esp32-idf4-20210202-v1.14.bin).

I hope that you could potentially fix this issue soon and that this information is helpful to you.

In any case, thank you very much for your efforts.

kiinging, 2021-10-17:
> Dear authors,
>
> I am experiencing exactly the same problems as depicted in this post. In short, when I use Thonny for running my simple test code (I am a beginner), the program runs fine; only when I [...]

I am experiencing the same problem. I am using Wemos Lolin32 which has a ESP-WROOM-32 microcontroller. I use Thonny.  I was not able to stop the code from running (Device is busy ..) using ctrl + c. Then, I tried to reinstall the software. I was also not able to install and update firmware, until I short pin GPIO0 to GND. So I shorted the pin GPIO0 to GND, and use Thonny to flash the ESP32. 
The problem is not gone yet. I disconnected GPIO0 to GND, I am still have the same problem as below:

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - use Stop/Restart to interrupt more and enter REPL.
 
I don't understand the ESP32 is already flashed with a new firmware. It supports to be empty and not busy.  Thonny also shows "
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.2s
Hard resetting via RTS pin...
Erasing done".

The last program that I tried  on this ESP32 is the deepsleep function in main.py.

Then, I tried to use Arduino IDE and install a simple Blink program to the ESP32. It works. The LED blink.

Then, I go back and reinstall the ESP with the micropython firmware. I found that I still need to short GPIO0 so that I can install the firmware. After that, the same problem. The device is still busy.

fyi, i hv three Wemos Lolin ESP32s that it do not need to short the GPIO0 pin in order to install the firmware. 

I hv run out of clues how to safe the ESP32?

aivarannamaa, 2021-11-04:
I've read reports about people not being able to interrupt deep sleep even when using simple and plain serial terminals for communicating with the MicroPython device.

One of the solutions is to start the main program with a few seconds of regular sleep, which would give you the time window for interrupting the program just after a hard reset.

jawii, 2021-11-16:
I found a hacky way of 

1. Upload the the circuitpython firmware uf2 
2. Upload pico firmware 
3. Now Thonny opened straight to REPL

Womper999, 2022-03-13:
Hi there,

I'm new here and want to program the pico.

I had a problem with the main.py

Now Thonny always gets the message:

Device is busy or does not respond. Your options:

   - wait until it completes current work;
   - use Ctrl+C to interrupt current work;
   - use Stop/Restart to interrupt more and enter REPL.

although no device is connected.
How can I reset Thonny?
I've also reinstalled Thonny to no avail

codiak6335, 2023-09-05:
When used with a lilygo T-DisplayS3 the stop botton causes a soft reboot and boot.py restarts on cycle.
A ctrl+c is required to stop the actual run time.