thonny/thonny #1583: Install From Local File < Manage Packages < Thonny

matthewfelgate2, 2021-01-30:
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pip_gui.py", line 655, in _handle_install_file_click
    filename = askopenfilename(
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1920, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1960, in askopenfilename
    args = cls._convert_common_options("Open file", **options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1994, in _convert_common_options
    filename = _options_to_zenity_filename(options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2036, in _options_to_zenity_filename
    return options["initialdir"] + os.path.sep
TypeError: unsupported operand type(s) for +: 'method' and 'str'

```

aivarannamaa, 2021-01-30:
Thank you for the report!

This bug should be fixed in Thonny 3.3. You are using an older version, right?

In case you were trying to install plug-in for RPi Pico: this plug-in doesn't work in Thonny 3.2 and is not required in Thonny 3.3.3 and above. It was supposed to be a temporary plug-in during development and got mentioned in Pico's public docs by mistake.

aivarannamaa, 2021-01-30:
Most Linux package repos don't have Thonny 3.3 yet, so you need to either uninstall the package and switch to binary bundle at thonny.org:

```bash
bash <(wget -O - https://thonny.org/installer-for-linux)
```

... or use `MicroPython (generic)` back-end for accessing Pico (if it was the reason you tried to install a local whl)

closed 2021-02-17 by aivarannamaa