thonny/thonny #1584: Fails when trying to install local wheel

pbarada, 2021-01-30:
Thonny v 3.2.7 on Ubuntu-20.04 LTS (installed via "sudo apt-get install thonny")

Tools -> Manage plug-ins -> Install from local file fails with:

Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pip_gui.py", line 655, in _handle_install_file_click
    filename = askopenfilename(
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1920, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1960, in askopenfilename
    args = cls._convert_common_options("Open file", **options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1994, in _convert_common_options
    filename = _options_to_zenity_filename(options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2036, in _options_to_zenity_filename
    return options["initialdir"] + os.path.sep
TypeError: unsupported operand type(s) for +: 'method' and 'str'

pbarada, 2021-01-30:
Should have said when clicking on "here" under "Install from local file" causes the failure.

aivarannamaa, 2021-01-30:
Thank you for the report! Please see #1583

closed 2021-01-30 by aivarannamaa