thonny/thonny #1585: thonny 3.3.3: ImportError: libXft.so.2: cannot open shared object file: No such file or directory

2d4d, 2021-01-31:
got this error after upgrading to thonny 3.3.3 despite having libxft2 installed:

INFO    thonny: Thonny version: 3.3.3
ERROR:root:Internal launch or mainloop error
Traceback (most recent call last):
  File "/home/a/apps/thonny/lib/python3.7/site-packages/thonny/__init__.py", line 192, in launch
    from thonny import workbench
  File "/home/a/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 15, in <module>
    import tkinter as tk
  File "/home/a/apps/thonny/lib/python3.7/tkinter/__init__.py", line 36, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
ImportError: libXft.so.2: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/home/a/apps/thonny/lib/python3.7/site-packages/thonny/__init__.py", line 192, in launch
    from thonny import workbench
  File "/home/a/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 15, in <module>
    import tkinter as tk
  File "/home/a/apps/thonny/lib/python3.7/tkinter/__init__.py", line 36, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
ImportError: libXft.so.2: cannot open shared object file: No such file or directory

was fixed by:
sudo  apt-get install libxft2:i386 lib32ncurses5
from https://stackoverflow.com/questions/62718951/how-to-fix-libxft-so-2-cannot-open-shared-object-file-when-simulating-hardware

would be nice to get the hint during the installation.

aivarannamaa, 2021-01-31:
Thank you for the report!

You're having 32-bit Linux, right? Which distro and which version?

2d4d, 2021-01-31:
arg, sorry, grabbed thonny-3.3.3-i686.tar.gz on x64, will close this.

closed 2021-01-31 by 2d4d

aivarannamaa, 2021-01-31:
Great, that was easy :) 

I recommend using the published bash script, which will download the correct version.