thonny/thonny #1586: Pi pico multitasking causes Thonny Management Error

mlewus, 2021-01-31:
I tried to do a simple multitasking example on my spanking new Pi Pico, using the second core to flash a LED. When I run it, the code works as expected but Thonny generates a Management Error. The pico is attached via usb to a MacBook Pro running MacOS 10.15.7, and the latest release version of Thonny 3.3.3.

The code:
```
import machine
from machine import Pin
import time
import _thread

def ltask(blinks,delay):
    led = Pin(25, Pin.OUT)
    for i in range(blinks):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print("CPU1 Done")

_thread.start_new_thread(ltask, (20, 0.2))
```
Running it has the expected result on th pico, the LED blinks at 2.5Hz for 20 cycles. But Thonny pops up a Management Error box and a long list of error messages. The complete output is:

```>>> %Run -c $EDITOR_CONTENT
THONNY FAILED TO EXECUTE COMMAND Run

SCRIPT:

class __thonny_helper:
    try:
        import uos as os
    except ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = dict()
    last_repl_values = []
    @classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.last_repl_values.append(obj)
            cls.last_repl_values = cls.last_repl_values[-5:]
            print('[object_link_for_thonny]', cls.repr(obj), '@', id(obj), '[/object_link_for_thonny]', sep='')

    @staticmethod
    def print_mgmt_value(obj):
        print('<thonny>', repr(obj), '</thonny>', sep='', end='')

    @staticmethod
    def repr(obj):
        try:
            s = repr(obj)
            if len(s) > 1000:
                s = s[:1000] + "..."
            return s
        except Exception as e:
            return "<could not serialize: " + str(e) + ">"

    @classmethod
    def listdir(cls, x):
        if hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @classmethod
    def getcwd(cls):
        if hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 6
SyntaxError: invalid syntax

ManagementError
>>> THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>OKTraceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax

CPU1 Done
```

You can see that after the error, the REPL prints "CPU1 Done" which is the print statement in my code, completing normally. 

If I run the same test, but replace `_thread.start_new_thread(ltask, (20, 0.2))` with `task(20, 0.2)` to run the code on the default core, Thonny does not generate an error message.

I am relatively new to Python and can't make any sense of the error output. Any suggestions on how to proceed?

aivarannamaa, 2021-02-01:
Thank you for the report!

Unfortunately there is no good solution to this problem, as there is single communication channel for the output of your program and output from Thonny's hidden management commands. If you are printing only in the main thread, then these actions don't disturb each other. Problem occurs when a secondary thread keeps running (and outputting) after the main thread returns to the prompt (and Thonny thinks the device is therefore idle).

In CPython I'd recommend joining the threads in the end of your script (ie. making sure the program doesn't return to the prompt before all threads are finished), but I don't know whether MicroPython has means for this (from this I suspect that it doesn't: https://stackoverflow.com/a/56359905).

I guess I need to make the output parsing smarter.

mlewus, 2021-02-01:
@aivarannamaa Thanks For the quick reply. I read the stack overflow post and I might try to implement _is_alive() locally. Or find another way to ensure the second core finishes before returning to the REPL.

mlewus, 2021-02-02:
I confirmed the behavior of multitasking. For instance:
```
import machine
from machine import Pin
import time
import _thread

def ltask(blinks,delay):
    led = Pin(25, Pin.OUT)
    for i in range(blinks):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print("Ended thread 2")

print("Started thread 1")
_thread.start_new_thread(ltask, (20, 0.2))
# thread will terminate in ~ 8 seconds
print("Started thread 2")
time.sleep(10)
print("Ended thread 1")

```
Results in:
```
>>> %Run -c $EDITOR_CONTENT
Start thread 1
Start thread 2
End thread 2
End thread 1
>>> 
```
As you noted, as long as the main thread on CPU 1 is still running when the thread on CPU 2 completes, all works as expected and thread on CPU 1 exits to the REPL.

I'm working on a Mac and I'm not sure how much effort it will take to build Micropython on this platform. If I can get that working I'll try to implement is_alive() based on your previous link. Either way, this is a suitable workaround for me, since my main thread will never terminate in normal operation (endless loop).

aivarannamaa, 2021-02-02:
> I'll try to implement is_alive()

The possibility to create and wait for a lock / mutex seems be even more useful. Have you considered making a feature request at https://github.com/micropython/micropython/issues/new ? 

I was asking around at MicroPython forum, but without much success https://forum.micropython.org/viewtopic.php?f=2&t=9705&p=54275

mlewus, 2021-02-02:
All I really need is a mutex so the main thread can tell when the cpu 2 thread is running. I tried this:
```
import machine
from machine import Pin
import time
import _thread

def ltask(blinks,delay):
#set mutex - mPython won't let you start more than one thread on Pic
# so no need to check value
    global a_mutex
    a_mutex = True
    led = Pin(25, Pin.OUT)
    for i in range(blinks):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print("Ended thread 2")
# release the mutex
    a_mutex = False
    return()

print("Started main thread on CPU 1")
a_mutex = False
_thread.start_new_thread(ltatsk, (20, 0.2))
print("Started concurrent thread on CPU 2")
#Wait for thread to terminate
print("Wait for concurrent thread to finish")
while a_mutex == True:
    pass
#time.sleep(10)
print("Concurrent thread on CPU 2 finished")
pass
print("Exit main thread to REPL on CPU 1")
```
Output:
```
>>> %Run -c $EDITOR_CONTENT
Started main thread on CPU 1
Started concurrent thread on CPU 2
Wait for concurrent thread to finish
Ended thread 2
Concurrent thread on CPU 2 finished
Exit main thread to REPL on CPU 1
>>> 
```
So it seems to work OK. There is also an example at [https://github.com/micropython/micropython/blob/master/tests/thread/thread_shared1.py](url) that shows how to share data structures between threads. This is sufficient for my needs.

One more comment, which almost certainly has the same root cause as the original issue. If you try to print from both cores at the same time, you get garbage. Thonny does not crash, but the output is garbled. If Thonny gets its input a character acan't imagine how you would go about fixing that,

aivarannamaa, 2021-02-03:
> If you try to print from both cores at the same time, you get garbage. Thonny does not crash, but the output is garbled.

I didn't expect this. I was told by a MicroPython developer that if you are doing a `print` with single argument, then the output should appear intact even if another thread is printing at the same. Seems that this is not the case still.

aivarannamaa, 2021-02-03:
I now noticed, that one of Thonny's hidden internal commands received a SyntaxError. This means that not only the output was disturbed but also input. It looks like I need to find a simple repeatable test case and report it under MicroPython project.

mlewus, 2021-02-03:
FWIW I noticed that it did not just send the characters from the print statements interspersed, but sent other "junk" as well.

aivarannamaa, 2021-02-03:
> FWIW I noticed that it did not just send the characters from the print statements interspersed, but sent other "junk" as well.

Good to know, thanks! If you can, then please post the output here as well, next time you see it.

mlewus, 2021-02-04:
Here is the code using my mutex to prevent CPU 1 from re-entering the REPL before CPU 2 is complete.
```
import machine
from machine import Pin
import time
import _thread

#blink the LED on the Pico
def ltask(blinks,delay):
   # set mutex True/False since mPython won't let you start more than one thread on RPi Pico
    print("Started concurrent thread on CPU 2")
    global a_mutex
    a_mutex = True
    led = Pin(25, Pin.OUT)
    for i in range(blinks):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print("Ended thread on CPU 2")
# release the mutex
    a_mutex = False
    return()

print("Started main thread on CPU 1")
a_mutex = False
_thread.start_new_thread(ltask, (20, 0.2))

#Wait for thread to terminate
print("Wait for concurrent thread to finish")
while a_mutex == True:
    pass
#time.sleep(10)
print("Main thread detected CPU 2 thread finished")
pass
print("Exit main thread to REPL on CPU 1")
```
This is what is printed:
```
>>> %Run -c $EDITOR_CONTENT
Started main thread on CPU 1
WSatiatrt feodr c oncocnurcurerrnetn tt hrtheraeda don  tCoPU f i2n
ish
Ended thread on CPU 2
Main thread detected CPU 2 thread finished
Exit main thread to REPL on CPU 1
>>> 
```
All I saw here was interspersed characters. Unfortunately I did not keep the original code, but it is possible I misinterpreted the output and there were no other characters in the output. I'll work with it some more and let you know if I discover any errant characters.

mlewus, 2021-02-04:
I spent more time with this and there are definitely no other characters being printed other than the interspersed print statements. Given the other inter-core comm limitations (see below) I am not sure it will be something they can fix. 

I came across some behavior in multicore tasking (_thread) in Micropython on the pico that might affect Thonny. Writes to globals are likely communicated through the FIFO structure via the C/++ SDK and don't get immediately communicated so you can't use a global variable as a mutex.  If you set a global in core 1 and change it in core 2, it takes about 1mS for the change to be reflected in core 1.

I looked at the multicore features in the RP2040 datasheet, C/C++ SDK and Python SDK. It appears there is no shared memory and Inter-core communication is limited to a FIFO mailbox. There is also a non-interrupt safe mutex using _thread,allocate_lock() that is suggested as a means to protect the SDK libs which, according to the Pi docs, are not re-entrant. 

This may not be relevant to you but I thought I'd communicate it in the event that it affects future changes to Thonny. 

I replaced the global mutex with thread.allocate_lock() and this works with no delay. I spent some time with varying versions of this code, printing text, variables, etc and in every case all I saw was interspersed characters.

FWIW here is the new test, which prints a number (pi) in task 1 while printing some text in task 2:
```
import time
import _thread
import math

def cpu2task():
    with lock:
        print("THEQUICKBROWNFOXJUMPSOVERTHELAZYDOGSBACK")
        # now do some stuff
        time.sleep(1)
        
print("Interspersed print test")
#wait to print first message
time.sleep(0.1)
#get the lock
lock = _thread.allocate_lock()
loops = 0
_thread.start_new_thread(cpu2task, ())
# this print arrives in pieces
print(math.pi)
while lock.locked() == True:
    loops += 1
print("Waited for",loops, "loops")
```
Output:
```
>>> %Run -c $EDITOR_CONTENT
Interspersed print test
T3.HE14Q1U5I9C3KB
ROWNFOXJUMPSOVERTHELAZYDOGSBACK
Waited for 58450 loops
>>> 
```

aivarannamaa, 2021-02-16:
I was able to reproduce this and it looks like MicroPython problem to me, so I filed https://github.com/micropython/micropython/issues/6899