thonny/thonny #1589: Automatic reload files after external changes

bulletmark, 2021-02-02:
I see that a reload prompt was added after issue #397. The OP there asked if it could be considered to make the reload automatic and the response was "I'll postpone adding automatic reload option until more users request this". Can this please be considered now?! Perhaps add an option so the default action is to prompt as present, but if the "auto reload files after external change" option is set then just reload all changed files on focus.

Actually, Visual Studio Code has always done this by default, and without requiring focus. I.e. if I edit a file in vim then when I write the file I see it update immediately in my adjacent vscode window. It works great with external editors and I'd personally prefer that approach.

I am on Arch Linux using standard package thonny 3.3.3.-1 with MicroPython on the Raspberry Pi Pico. Fixing this issue would make using thonny + vim more seamless.

aivarannamaa, 2021-02-02:
Thank you for the suggestion!

Do I understand correctly, that you just want means to get rid of the confirmation dialog?

bulletmark, 2021-02-02:
There is already a means, I can just press the return key because that popup has focus. But I (and the OP of that bug I referenced) don't want to see the popup at all. Thonny should just see the file has changed and re-read it immediately when Thonny gets focus again. We don't want to see or bother with that popup. That is what my 1st paragraph above is about.

My second paragraph above describes a more ambitious suggestion which I personally would prefer (and the OP of that bug did not discuss), i.e. please do exactly what vscode does. I.e. set file watchers (inotify?) on opened files and immediately update the corresponding window content when the files are changed externally, before Thonny even gets focus. As I said, vscode does this now and it works very well with external editors.

I am trying to improve the workflow to use vim with the new RPi Pico + MicroPython. I want to just save the file in vim, then just press Run in Thonny to have it downloaded and run. Maybe an "auto run on external file change, or save" would be even better(?!) as that way I could leave the Thonny window up to view errors etc, and just stay within vim.

aivarannamaa, 2021-02-02:
Thank you for the clarification!

I've always thought that file watching is hard to do in cross-platform apps, but the watchdog library seems to be a nice solution for this.

> I am trying to improve the workflow to use vim with the new RPi Pico + MicroPython. I want to just save the file in vim, then just press Run in Thonny to have it downloaded and run.

In this case you might want to use a command line tool for running code on or uploading to a MicroPython device:

* https://docs.micropython.org/en/latest/reference/pyboard.py.html
* https://github.com/scientifichackers/ampy
* https://github.com/dhylands/rshell

bulletmark, 2021-02-02:
Thanks for the links. Thonny (for UI) and rshell (for command line) are mentioned in the Pico docs. I had already tried rshell but it merely hangs on my Linux system and since there has been no release since May 2020 (in spite of the new Pico) I figured it is not seriously maintained. Pyboard.py is extremely simple and works very well for what I want. I am surprised there is no formal package (not even on PyPi) for it. You are expected to merely download that single source file from the main MicroPython repo. So updates would have to be manually managed. Ampy seems to give little more than pyboard.py so seems a bit overkill.

So I will use pyboard.py - but I still agree with the OP of issue #397 for an option to make Thonny automatically reload.

aivarannamaa, 2021-02-26:
Alright, I'll add the option to suppress the confirmation in Thonny 4.0

closed 2022-08-14 by aivarannamaa

OlegAlexander, 2023-12-05:
Hello and thank you for creating Thonny!! 

I'm using Thonny to teach my son Python. A common programming task is to save some data to a json or csv file from _within_ Thonny (internal file change, not external). Currently, the only way to reload a json/csv file is to close it and open it again. 

Is it possible to either:

- Automatically reload a file if it's been changed when the user clicks on that file's tab?
- Add a Reload option under the right-click menu for a file's tab next to Close, Close Others, and Close All, as well as the File menu?

Thank you! (My apologies for commenting on a closed issue. If you'd like, I can open a new one.)

aivarannamaa, 2023-12-05:
@OlegAlexander, I agree something needs to be done here. Please copy your text to a new issue, then I can schedule it for version 5.0.