thonny/thonny #1595: Problem with Polish keyboard layout

michalsieron, 2021-02-05:
I just discovered Thonny and so far I love it. There is just one, rather big blocker for me to use it. When I try to type in some Polish characters with `AltGr + <key>` I get wrong results. 

For example:
* `AltGr + a` should give `ą`, but gives `¹`
* `AltGr + c` should give `ć`, but gives `æ`
* and so on

I am on Windows 10 with __Polish (Programmers)__ keyboard layout using Thonny 3.3.3 Portable.

aivarannamaa, 2021-02-05:
Thank you for the report!

Could you please run following code:

```
import tkinter

root = tkinter.Tk()
text = tkinter.Text(root)
text.grid()
root.mainloop()
```

It should give you a minimal text editor which uses the same UI toolkit as Thonny (Tkinter).

Can you enter your characters in this editor?

michalsieron, 2021-02-05:
That was fast. Unfortunately it gives the same result

michalsieron, 2021-02-05:
Seems like someone had [similar issue](https://stackoverflow.com/questions/14472766/tkinter-determine-keyboard-layout). I don't think that detecting layout is necesary here. A way to change current layout in Options should suffice.

aivarannamaa, 2021-02-05:
I'll postpone this until releasing Thonny 4.0 with a newer Tk -- there is a chance that this is fixed in Tk 8.6.11.

aivarannamaa, 2023-07-05:
Thonny 4.1.1 now comes with Tk 8.6.13. Is this problem fixed for you?

michalsieron, 2023-07-09:
I don't use Windows anymore, but I decided to check it. Same behavior as before.
But, don't worry (or maybe do worry) as this is a Tk bug.
I spent multiple evenings looking around and trying things.

Short story is: to make it work, I need to change keyboard layout and then go back.

Almost the same issue was reported and then fixed in 2006 ([ticket 1374119](https://core.tcl-lang.org/tk/tktview?name=1374119fffffffff)).
So why do I face this problem? Wasn't this fixed?

You see, Windows uses two language IDs (actually they were deprecated ([Windows Language Code Identifier (LCID) Reference - section 1.3](https://winprotocoldoc.blob.core.windows.net/productionwindowsarchives/MS-LCID/%5bMS-LCID%5d.pdf)), but Tk still uses them).
Together, they create one big language identifier ([Language Identifiers](https://learn.microsoft.com/en-us/windows/win32/intl/language-identifiers)).

One language ID represents a primary language, while the second one is a sublanguage identifier.

In my case I use **English (United Kingdom)** as primary language (it doesn't have to match system language(!), although in my case it does) and **Polish (Programmers)** keyboard layout, which sets sublanguage to **Polish**.
[image]

The fix I mentioned earlier uses primary language to set inital encoding info ([win/tkWinX.c#L253-L258](https://github.com/tcltk/tk/blob/e7c8688438f4de50bacba0220fb7a4fb9b5fd0ed/win/tkWinX.c#L253-L258)). And it worked, because I presume the user had the same primary and sublanguage set.

HOWEVER, the event handler for `WM_INPUTLANGCHANGE` ([win/tkWinX.c#L740-L743](https://github.com/tcltk/tk/blob/e7c8688438f4de50bacba0220fb7a4fb9b5fd0ed/win/tkWinX.c#L740-L743)) uses some obscure value I have no idea where it comes from, but from my experiments with different languages and keyboard layouts, corresponds to a value one could derieve from sublanguage ID.

To fix this, would probably mean to change logic setting initial encoding info, to use sublanguage ID instead of the primary language ID.

After all this, I realized that I should probably search Tk bugtracker for this issue.
I found [ticket a9bd2ab98bdedf4420f6](https://core.tcl-lang.org/tk/tktview/a9bd2ab98bdedf4420f6) from 2017, which is kinda similar and shows that there are more issues regarding keyboard layouts in Tk for Windows.
However, there is no recent activity, so that's sad.

When I find some more time, maybe I will search the bugtracker a bit more and describe this issue there as well.
But for now, this issue doesn't seem to be getting fixed any time soon.

aivarannamaa, 2023-08-05:
Sorry for the late reply!

Thank you for this investigation! It's a pity Tk is not more popular, but at least it is not dead yet and some bugs get fixed from time to time.

andrus007, 2026-02-13:
@aivarannamaa - just to update: still the same issue in Thonny 4.1.7 (Tk 8.6,13), on Windows 11....
(but at least the workaround from @michalsieron helps temporarily )