thonny/thonny #1601: Thonny doesnt want to work

Jawanen, 2021-02-07:
Why this isnt working?

`Traceback (most recent call last):
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1956, in toolbar_handler
    handler(*args)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 471, in cmd_stop_restart
    self.restart_backend(True)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 602, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 270, in __init__
    super().__init__(clean, get_interpreter_for_subprocess(executable))
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 38, in __init__
    super().__init__(clean, executable)
  File "C:\Users\Jaakko\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 866, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\Jaakko\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 933, in _start_background_process
    **extra_params
  File "C:\Users\Jaakko\AppData\Local\Programs\Thonny\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Jaakko\AppData\Local\Programs\Thonny\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied`

Jawanen, 2021-02-07:
Now i deleted all files and it isnt working

aivarannamaa, 2021-02-08:
Thank you for the report!

Have you been able to use Thonny in the same computer before? If yes, then have you changed something in Thonny?

Jawanen, 2021-02-20:
yes

Jawanen, 2021-02-20:
i think i repaired it

Jawanen, 2021-02-20:
now i am tryin to repair it

aivarannamaa, 2021-02-26:
Have you solved the problem? What did you do?

closed 2021-03-06 by aivarannamaa

Phantom061, 2023-05-06:
@Jawanen what did you do? I have the same issue and I am not able to fix it