thonny/thonny #1603: Sync MicroPython clock to local time by default

aivarannamaa, 2021-02-07:


closed 2021-02-07 by aivarannamaa

captn3m0, 2021-02-20:
@aivarannamaa While this is cool, this is confusing behavior, since this isn't documented. I was confused for a while (how is my Pi reporting the correct time? Is this code even running on the Pi?) before I managed to track down the code.

I've added it in [the FAQ for now](https://github.com/thonny/thonny/wiki/FAQ), but you might wanna document this better.

aivarannamaa, 2021-02-20:
Thanks! 

Do yo think you could move this information to https://github.com/thonny/thonny/wiki/MicroPython instead?

aivarannamaa, 2021-02-20:
Thanks again! It's much better now. I changed your last sentence, though:

> Until version 3.3.3 RTC was synced to UTC. Since 3.3.4 it is synced to local time by default, but you can choose UTC by setting ``utc_clock`` to ``True``.

Also, I added an issue for adding a dropbox for this configuration option to MicroPython back-end configuration dialog (#1638).

captn3m0, 2021-02-21:
Having it in the settings makes so much sense. Thanks 🎉