thonny/thonny #1605: portable pip installs to user directory

thouseweart, 2021-02-09:
Quite possibly this is a case of user error.

I made a portable miniconda installation on a shared network drive, and used this portable pip to install thonny, without the "--user" flag. 

After copying the portable python installation (including Thonny) to another location on the shared drive, I was surprised to see that the new Thonny was still using the old python installation.

The reason appears to be that pip installed some Thonny info in "C:\Users\[username]\AppData\Roaming\Thonny". In this folder, configuration.ini contains a hardcoded "working_directory" path, which points to the original python install.

Is there a good way to make completely isolated/sandboxed Thonny installs using pip? Ideally, I'd like to store nothing in the user profile.

As a side note, "pip uninstall thonny" doesn't seem to remove anything in AppData\Roaming\Thonny. I feel like this is undesirable behavior, but I'm also not super familiar with pip/conda idioms.

Thanks in advance for any input or help!

thouseweart, 2021-02-09:
Version info:

- Thonny 3.3.3
- Windows 10
- Python 3.8.5
- Pip 20.2.4
- Conda 4.9.2

thouseweart, 2021-02-09:
In Thonny GUI, my "Tools > program folder" is correct, but "Tools > data folder" is incorrect/user-specific.

It looks like as a workaround I can delete configuration.ini in the data folder, and write its contents to defaults.ini in the program folder?

Not an ideal fix, but it'll work for me if there's not a better option 😊

thouseweart, 2021-02-09:
I deleted the "data folder" as a test, and on the next startup of Thonny, this folder was recreated under AppData\Roaming.

So it does seem to be a Thonny behavior rather than a pip behavior.

I know that Thonny offers a portable build, is it possible to switch Thonny to "portable mode" after it's installed, or to ask pip to build it as portable using "--install-option" flags?

aivarannamaa, 2021-02-09:
AppData\Roaming\Thonny is indeed maintained by Thonny, as you already discovered.

Currently the portable mode assumes you're using Thonny+Python bundle, but you could try creating an empty marker file named *portable_thonny.ini* next to your python.exe

thouseweart, 2021-02-09:
This worked! Creating the _portable_thonny.ini_ in my portable miniconda folder and running Thonny gave me the first-run splashscreen with "Welcome to Thonny [portable]" in the titlebar. Thonny also created a new, local "data folder" in miniconda\user_data and populated it with a _configuration.ini_ and logfiles. Thonny titlebar also now reads "Portable Thonny".

Thanks so much for responding so quickly and for helping me out!!

closed 2021-02-09 by aivarannamaa

thouseweart, 2021-02-11:
A new wrinkle! After copying this portable conda/thonny to another directory and running Thonny via the executable (miniconda\Scripts\thonny.exe), Thonny somehow uses the program folder and data folder from the previous conda/thonny install. This continues to happen even after deleting the new "miniconda\user_data directory".

However, I tried launching via this simple python script:
```
import thonny
print(f"THONNY USER DIR: {thonny._compute_thonny_user_dir()}")
print(f"IS PORTABLE? {thonny.is_portable()}")
print("LAUNCHING!")
thonny.launch()
```
and it works like a charm, generating the user_data directory because I had deleted it, printing out the correct user dir, and using the correct data folder and program folder once it starts up.

What's the difference between starting with the .exe vs. thonny.launch(), and how does the .exe method remember my previous portable python install?

Thanks again!

aivarannamaa, 2021-02-11:
> What's the difference between starting with the .exe vs. thonny.launch(), and how does the .exe method remember my previous portable python install?

This is a `setuptools` (library for making pip-installable package distrubutions) thing -- all executables in the Scripts directory are generated at `pip install` time and become tied to the absolute path of the main interpreter (or virtual environment environment)  running pip.

Other than that, the exe does exactly what you wrote in your script. An alternative to this script is calling `python -m thonny` (which ends up calling this module: https://github.com/thonny/thonny/blob/master/thonny/__main__.py)

thouseweart, 2021-02-11:
Oh, good to know! `python -m thonny` works well for me, thank you!