thonny/thonny #1607: The Debug icons are grayed out

simcha07, 2021-02-10:
I have Thonny installed on windows-10.
It is working well, except the debugger.
All debug icons are grayed out. 
How come?
(A  screenshot is attached.)
[image]

simcha07, 2021-02-10:
Found (the) (a) solution!
I installed the Thonny IDE again but now not in my own language but I chose for "English".
Now the debug icon(s) are available and not grayed out anymore.
Feel free to respond if there is a solution for an Thonny installation in my own language.
Thanks in advance.

aivarannamaa, 2021-02-10:
From the screenshot I understand that you are using MicroPython back-end. Debugging is not supported in MicroPython.

closed 2021-02-10 by aivarannamaa

simcha07, 2021-02-11:
> From the screenshot I understand that you are using MicroPython back-end. Debugging is not supported in MicroPython.

That's a pity, thanks for responding!