thonny/thonny #1611: module 'tabulate' not found

RNeuwirth, 2021-02-14:
In my python (version 3.9.1) application "from tabulate import tabulate" prompts" ModuleNotFoundError: No module named 'tabulate'". IDLE Shell 3.9.1 runs without a problem.

aivarannamaa, 2021-02-14:
Most probably your IDLE and Thonny are using different Python interpreters, one of them having `tabulate` package and the other not. 

If you want to use it in Thonny then either make Thonny use the same interpreter as IDLE (Run => Select interpreter => Alternative Python 3 ...) or install `tabulate` for Thonny's interpreter as well (Tools => Manage packages).

RNeuwirth, 2021-02-14:
> <reply@reply.github.com>
> <author@noreply.github.com>

Thank you.

[2]. 
-------------------------
Email sent using Optus Webmail

Links:
------
[1]
https://github.com/thonny/thonny/issues/1611#issuecomment-778736769
[2]

closed 2021-02-14 by aivarannamaa