thonny/thonny #1612: New editor feature: UPPERCASE, lowercase and Capitalize selected words

IhorNehrutsa, 2021-02-14:
For example, the words **"qwerty and asdfgh"** are selected in Thonny IDE.
When Ctrl+[ are pressed they will convert to  UPPERCASE: **"QWERTY AND ASDFGH"**
When Ctrl+] are pressed they will convert back to  lowercase: **"qwerty and asdfgh"**
When Ctrl+| are pressed they will capitalize the first letter in the words: **"Qwerty And Asdfgh"**

This editor feature will useful for CONSTANT/variables/Classes/function naming.
Thanks

aivarannamaa, 2021-02-14:
Thank you for the suggestion, but I don't think many users would find it useful. 

If I understand correctly, it could be useful only if you used wrong case initially and want to change your mind later. I think a more general "Rename all occurrences of an identifier" would be much more useful in this case.

IhorNehrutsa, 2021-02-15:
Yes, this feature helps correct the wrong case right now.
Sometimes I type **PRINT(** or something similar when I accidentally pressed CapsLock.

I have an expirience with an editor with this feature, it was useful. Try and I think you like it.

IhorNehrutsa, 2021-02-17:
A small example from today's case:
We have a constant and a function which uses the global constant.
```
PARAMETER = 'some constant'
...
def some_func():  
    do something with PARAMETER
...
some_func()
```
When refactoring, you need to change some_func  which will use the global constant as the argument.
```
PARAMETER = 'some constant'
...
def some_func(parameter):  
    do something with parameter
...
some_func(PARAMETER)
```
This feature helps to make it easier and faster: one Ctrl+] and several Copy-Past.
Thanks.

aivarannamaa, 2021-02-26:
Sorry, I'm still not convinced that it is useful feature for many users. But I created an issue for introducing "Rename" feature: #1653

closed 2021-02-26 by aivarannamaa

IhorNehrutsa, 2021-02-26:
The "Rename" feature is too heavyweight in this case.
The "UPPERCASE, lowercase and Capitalize" works here and right now under the editor cursor(without additional message boxes). 
It helps keep the flow of thoughts. (I think You know this state.)

https://en.wikipedia.org/wiki/Flow_(psychology)
https://web.archive.org/web/20181116215757/https://latysheva.ru/blog/flow-and-mihaly-csikszentmihalyi
https://medium.com/@bazzy/%D0%BA%D0%BE%D0%B3%D0%B4%D0%B0-%D0%B2%D0%BE%D0%B7%D0%BD%D0%B8%D0%BA%D0%B0%D0%B5%D1%82-%D1%81%D0%BE%D1%81%D1%82%D0%BE%D1%8F%D0%BD%D0%B8%D0%B5-%D0%BF%D0%BE%D1%82%D0%BE%D0%BA%D0%B0-709d55c11e90

houmie, 2025-02-15:
@aivarannamaa 

Pycharm allows this feature.  It's very useful.   Please reconsider it.  
In Pycharm I can select a whole row of currencies that is written in lowercase, and with a keyboard command I turn it all to uppercase.