thonny/thonny #1613: WebREPL Connection lost (EOF) in Thonny 3.3.3 ESP8266 MicroPython v.1.14

aziubin, 2021-02-14:
I am using ESP8266 running the latest MicroPython v1.14. I have configured WebREPL with password and can connect to it with the "official" WebREPL client for MicroPython (webrepl.html). The same does not work in Thonny, I have the following messages, trying to connect, then Stop/Restart backend couple of times as suggested by first error message, also tried to turn on/off ESP8266 and restart Thonny:

```
MicroPython v1.14 on 2021-02-02; ESP module with ESP8266

Type "help()" for more information.
>>> 
-----------------------------------------------------------------------------------------------------------

WebREPL connected

Connection lost (EOF)

Use Stop/Restart to reconnect.

-----------------------------------------------------------------------------------------------------------
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 827, in transfer_data
    message = await self.read_message()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 910, in read_message
    return frame.data.decode("utf-8") if text else frame.data
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Program Files (x86)\Thonny\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 54, in _wrap_ws_main
    loop.run_until_complete(self._ws_main())
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 66, in _ws_main
    await asyncio.gather(self._ws_keep_reading(), self._ws_keep_writing())
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 123, in _ws_keep_writing
    await self._ws.send(payload)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 555, in send
    await self.ensure_open()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 812, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedError: code = 1006 (connection closed abnormally [internal]), no reason

-----------------------------------------------------------------------------------------------------------------
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 827, in transfer_data
    message = await self.read_message()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 910, in read_message
    return frame.data.decode("utf-8") if text else frame.data
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Program Files (x86)\Thonny\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 54, in _wrap_ws_main
    loop.run_until_complete(self._ws_main())
  File "C:\Program Files (x86)\Thonny\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 66, in _ws_main
    await asyncio.gather(self._ws_keep_reading(), self._ws_keep_writing())
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 123, in _ws_keep_writing
    await self._ws.send(payload)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 555, in send
    await self.ensure_open()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\websockets\protocol.py", line 812, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedError: code = 1006 (connection closed abnormally [internal]), no reason

-----------------------------------------------------------------------------------------------------------------
```

My environment: Windows 10 Home, Thonny 3.3.3, Python 3.7.9 (32 bit) Tk 8.6.9
COM6 connection in Thonny works as expected, I can connect to ESP8266, run Python code and upload files, but I would like to use WebREPL if possible.

[image]

aivarannamaa, 2021-02-14:
Thank you for the report!

Which ESP8266 board are you using?

aziubin, 2021-02-14:
I have "Wireless module NodeMcu v3 CH340 Lua WIFI Internet of Things development board ESP8266" (see below). It works OK, I did not have any issues with it. I also tried another, Wemos D1 board, which had MicroPython 1.12 and the issue was not reproducible there and it was possible to use WebREPL in Thonny. Once I have upgraded the firmware on Wemos D1 board from MicroPython 1.12 to 1.14, WebREPL from Thonny stopped working alhtough the 'official' webepl.html still works. This, it looks like MicroPython 1.14 (or 1.13) specific issue if I noticed everything correctly. Pleae note that I verified the behaviour with two scenarios: when main.py runs some code on startup of the board and without this auto-start script.
[image]

seganku, 2021-02-15:
I am also seeing this issue with MicroPython 1.14 on 3 of my devices (ESP-01 and 2x ESP32).

I did also see an additional error, but I am not having success in duplicating it .. or I forget the exact sequence of events to generate it.  The jist of the error was that it was expecting `b'\r\n'`, but got `b''`

Thank you so much for all your work on this amazing set of tools!

seganku, 2021-02-15:
I have been testing mostly with the remote WebREPL webpage, not Thonny's internal WebREPL interpreter.

Testing now with MicroPython 1.13, and I found that error I'd "lost":
```
>>> 
──────────────────────────────────────────────────────────────────────────────

WebREPL connected
>>> help()
WARNING:root:Unexpected echo. Expected b'help()\r\n', got b'<thonny>'
MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32

Type "help()" for more information.
>>> 
```
Thonny 3.3.3

MicroPython 1.13 works "better" with Thonny 3.3.3, in that WebREPL can see what is typed on the serial, but I can't type in the WebREPL while Thonny is connected.  I do not have this problem if I connect to the serial port directly with PuTTY.  With PuTTY, I am able to type in either terminal window and the text/output will show up in both.

Thonny 3.3.3 also gives quite a lot of extra spam on the WebREPL:
```
raw REPL; CTRL-B to exit                                                                                                                     
>raw REPL; CTRL-B to exit                                                                                                                    
>                                                                                                                                            
MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32                                                                                     
Type "help()" for more information.                                                                                                          
>>>                                                                                                                                          
raw REPL; CTRL-B to exit                                                                                                                     
>OK>OK<thonny>'/'</thonny>>OK<thonny>['', '/lib']</thonny>>OK__main__          gc                uasyncio/stream   upip_utarfile             
_boot             inisetup          ubinascii         upysh                                                                                  
_onewire          machine           ubluetooth        urandom                                                                                
_thread           math              ucollections      ure                                                                                    
_uasyncio         micropython       ucryptolib        urequests                                                                              
_webrepl          neopixel          uctypes           uselect                                                                                
apa106            network           uerrno            usocket                                                                                
btree             ntptime           uhashlib          ussl                                                                                   
builtins          onewire           uhashlib          ustruct                                                                                
cmath             sys               uheapq            utime                                                                                  
dht               uarray            uio               utimeq                                                                                 
ds18x20           uasyncio/__init__ ujson             uwebsocket                                                                             
esp               uasyncio/core     umqtt/robust      uzlib                                                                                  
esp32             uasyncio/event    umqtt/simple      webrepl                                                                                
flashbdev         uasyncio/funcs    uos               webrepl_setup                                                                          
framebuf          uasyncio/lock     upip              websocket_helper                                                                       
Plus any modules on the filesystem                                                                                                           
>OK<thonny>(2029, 12, 31, 0, 0, 0, 0, 365)</thonny>>OK<thonny>True</thonny>>OK<thonny>(2021, 2, 15, 19, 17, 16)</thonny>>OK<thonny>{'boot.py'
: (32768, 0, 0, 0, 0, 0, 252, 666731347, 666731347, 666731347), 'network_config.py': (32768, 0, 0, 0, 0, 0, 650, 666730712, 666730712, 666730
712), 'webrepl_cfg.py': (32768, 0, 0, 0, 0, 0, 18, 666730678, 666730678, 666730678)}</thonny>>OK<thonny>{'reset': ('<function reset at 0x3ffe
5ad0>', 1073634000), 'gc': ("<module 'gc'>", 1061201904), 'bdev': ('<Partition type=1, subtype=129, address=2097152, size=2097152, label=vfs,
 encrypted=0>', 1073630896), 'ifconfig': ('<function ifconfig at 0x3ffe54a0>', 1073632416), 'uos': ("<module 'uos'>", 1061251460), 'webrepl':
 ("<module 'webrepl' from 'webrepl.py'>", 1073630640), 'do_connect': ('<function do_connect at 0x3ffe52f0>', 1073631984)}</thonny>>
```

Thonny also does not appear to leave the REPL session in a usable state after disconnect.  If I disconnect Thonny, then connect with PuTTY, I am unable to get to a REPL prompt.  REPL shows up properly after resetting the MicroPython 1.13.

Going back to MicroPython 1.14, WebREPL works just fine as long as I don't let Thonny 3.3.3 touch it, or reset between Thonny and WebREPL.

aivarannamaa, 2021-02-16:
Thank you for the input!

I was able to reproduce it and found the reason. MicroPython 1.14 supports a new and more efficient communication mode, which Thonny 3.3 tries to use. Unfortunately, WebREPL is sending some integer data in text mode and this causes the UnicodeDecodeError in your stacktraces. This also explains why the original WebREPL works -- you are using normal REPL mode there. 

I implemented a work-around so that when communicating over WebREPL, Thonny does not attempt to use the new mode and always uses the regular paste mode instead.

> Thonny 3.3.3 also gives quite a lot of extra spam on the WebREPL

Right, there is quite a lot going on behind the scenes, in order to support the file browser, variables table etc.

> Thonny also does not appear to leave the REPL session in a usable state after disconnect. If I disconnect Thonny, then connect with PuTTY, I am unable to get to a REPL prompt.

The reason is that Thonny uses raw REPL and this mode sticks even when you disconnect from Thonny. Press Ctrl+B in Putty to return to regular REPL.

> I can't type in the WebREPL while Thonny is connected.

Again, it's the raw REPL, which doesn't echo typed characters.

closed 2021-02-16 by aivarannamaa

aziubin, 2021-02-16:
Hello, thanks a lot for so fast response!
I can see that the fix is correction of thonny/plugins/micropython/bare_metal_backend.py
Can we try it now, just replacing bare_metal_backend.py which we have already installed, not waiting for the new Thonny build?
Also it would be great to have advanced mode if possible. Can you please send references to the documentation or to the code, so I could look? It might be a bug on Micro Python side too and we can open an issue against it if necessary.
Thanks!

aivarannamaa, 2021-02-16:
@aziubin, copying bare_metal_backend.py over the one in Thonny 3.3.3 might work. If not, then overwrite everything in `thonny` directory with current master.

The new raw-paste mode is described here: https://docs.micropython.org/en/latest/reference/repl.html#raw-mode-and-raw-paste-mode

I already filed the issue at MicroPython tracker: https://github.com/micropython/micropython/issues/6912

aziubin, 2021-02-20:
Verified that Thonny 3.3.4 can open MicroPython v.1.14 WebRepl (ESP8266).

seganku, 2021-02-20:
It appears that Thonny 3.3.4 + MicroPython 1.14 are still having problems when a remote WebREPL connects:

```
ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in _handle_normal_command
    response = handler(cmd)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 636, in _cmd_get_globals
    "{name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')}"
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 549, in _evaluate
    out, err = self._execute(script, capture_output=True)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 505, in _execute
    self._execute_with_consumer(script, consume_output)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 708, in _execute_with_consumer
    self._submit_code(script)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 567, in _submit_code
    self._submit_code_via_raw_paste_mode(to_be_sent)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 657, in _submit_code_via_raw_paste_mode
    self._raw_paste_write(script_bytes)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 691, in _raw_paste_write
    raise AssertionError("Unexpected read during raw paste: {}".format(data))
AssertionError: Unexpected read during raw paste: bytearray(b'p')
term: Exception in write() method, deactivating: OSError: [Errno 104] ECONNRESET
�
```

I see this issue is now closed, so maybe I'll start a new thread.  I was kinda thinking these issues were probably somewhat related.