thonny/thonny #1615: I get this error, all of a sudden

Idahowalker, 2021-02-15:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mysql/connector/network.py", line 39, in <module>
    from . import constants, errors
ImportError: cannot import name 'constants' from 'thonny.plugins.cpython' (/usr/lib/python3/dist-packages/thonny/plugins/cpython/__init__.py)

Tried reinstall. No work.

Tried complete remove and then install, no work

Idahowalker, 2021-02-15:
More info:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1707, in event_generate
    handler(event)
  File "/usr/lib/python3/dist-packages/thonny/assistance.py", line 138, in handle_toplevel_response
    msg["filename"], source, _get_imported_user_files(msg["filename"], source)
  File "/usr/lib/python3/dist-packages/thonny/assistance.py", line 943, in _get_imported_user_files
    for file in {name + ext for ext in [".py", ".pyw"] for name in module_names}:
  File "/usr/lib/python3/dist-packages/thonny/assistance.py", line 943, in <setcomp>
    for file in {name + ext for ext in [".py", ".pyw"] for name in module_names}:
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'

3rd, remove completely, reboot, reinstall, reboot, try, fail.

aivarannamaa, 2021-02-15:
When do you get this error? When starting Thonny or when running your program?

Have you used mysql.connector in your programs?

Idahowalker, 2021-02-15:
My error. I an error, once. see post 1.

reinstalled but forgot to install the librar. All is well now. sorry to had bothered.

closed 2021-02-15 by Idahowalker