thonny/thonny #1618: "ssl certificate expired" trying to install Micropython firmware

1hko, 2021-02-16:
I was following along with [this video](https://www.youtube.com/watch?v=_ouzuI_ZPLs) and tried to update my firmware using Thonny. When I clicked `Install`, my IDE attempts to download micropython but it fails with a "certificate expired" error.

To get around it for now, I just installed Micropython manually from the [Raspberry Pi Pico Getting Started](https://www.raspberrypi.org/documentation/pico/getting-started/) page. After launching Thonny, I was automatically connected and could run commands in the REPL and run programs using "Run current Script", `F5`.

Is the SSL cert something I can fix on my end?

aivarannamaa, 2021-02-16:
Thank you for the report!

I can't reproduce this in Ubuntu. Maybe it was a temporary problem with GitHub's or micropython.org's certificates.

Do you still get this error? If yes, then which platform are you using?

1hko, 2021-02-16:
Thanks for the quick reply! I am currently using Windows 10. I will be able to try again later this evening.

1hko, 2021-02-16:
Still happening on Thonny 3.3.3, Windows 10 (64-bit), Python 3.7.9 (32-bit), Tk 8.6.9.

Thanks for your support. Here's the error I'm seeing

```none
Downloading 497664 bytes from https://micropython.org/resources/firmware/rp2-pico-20210202-v1.14.uf2

Traceback (most recent call last):
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 1350, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1277, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1323, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 972, in send
    self.connect()
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1447, in connect
    server_hostname=server_hostname)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\ssl.py", line 423, in wrap_socket
    session=session
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\ssl.py", line 870, in _create
    self.do_handshake()
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\ssl.py", line 1139, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 285, in _perform_work
    self._download_to_the_device(download_url, size, target_dir)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 347, in _download_to_the_device
    with urlopen(req, timeout=5) as fsrc:
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 543, in _open
    '_open', req)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 503, in _call_chain
    result = func(*args)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 1393, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 1352, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)>
```

aivarannamaa, 2021-02-16:
That's really strange. In my Windows 10 machine I can download it without problems. Can you download from this URL directly: https://micropython.org/resources/firmware/rp2-pico-20210202-v1.14.uf2 ?

1hko, 2021-02-16:
I thought it strange too. Yes, I am able to download that file using Firefox 85.0.2

1hko, 2021-02-17:
I tried uninstalling and reinstalling Thonny 3.3.3 again tonight and I'm experiencing the same error. I will continue with manual installs for now

aivarannamaa, 2021-02-17:
I'm closing this for now. Next major Thonny version will contain newer Python version (3.9). Hopefully this will fix your problem.

closed 2021-02-17 by aivarannamaa

ESAKKIMUTHU2127, 2021-10-05:
> Still happening on Thonny 3.3.3, Windows 10 (64-bit), Python 3.7.9 (32-bit), Tk 8.6.9.
> 
> Thanks for your support. Here's the error I'm seeing
> 
> ```
> Downloading 497664 bytes from https://micropython.org/resources/firmware/rp2-pico-20210202-v1.14.uf2
> 
> Traceback (most recent call last):
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 1350, in do_open
>     encode_chunked=req.has_header('Transfer-encoding'))
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1277, in request
>     self._send_request(method, url, body, headers, encode_chunked)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1323, in _send_request
>     self.endheaders(body, encode_chunked=encode_chunked)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1272, in endheaders
>     self._send_output(message_body, encode_chunked=encode_chunked)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1032, in _send_output
>     self.send(msg)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 972, in send
>     self.connect()
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\http\client.py", line 1447, in connect
>     server_hostname=server_hostname)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\ssl.py", line 423, in wrap_socket
>     session=session
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\ssl.py", line 870, in _create
>     self.do_handshake()
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\ssl.py", line 1139, in do_handshake
>     self._sslobj.do_handshake()
> ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 285, in _perform_work
>     self._download_to_the_device(download_url, size, target_dir)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 347, in _download_to_the_device
>     with urlopen(req, timeout=5) as fsrc:
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 222, in urlopen
>     return opener.open(url, data, timeout)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 525, in open
>     response = self._open(req, data)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 543, in _open
>     '_open', req)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 503, in _call_chain
>     result = func(*args)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 1393, in https_open
>     context=self._context, check_hostname=self._check_hostname)
>   File "C:\Users\me\AppData\Local\Programs\Thonny\lib\urllib\request.py", line 1352, in do_open
>     raise URLError(err)
> urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)>
> ```

I too have the same error anyone pls help me to  resolve it

aivarannamaa, 2021-10-05:
Please see #1986

SoyaSource, 2022-02-25:
The issue with Thonny is still occuring (as of today) on v3.3.13.

I followed aivarannamaa's advice and it worked to shift the IDE into recognising the Pico!
- install directly by downloading the file as per the link aivarannamaa gave.
- copy the file from Downloads over to the USB folder for the Pico
- the Pico will automatically stop being USB recognisable, but will come up in Thonny.

clark77449, 2022-04-06:
I am having the same issue. Site Certificate Expired Error. I tried updating the certificate and using a PIP command in DOS to update the certificate. I had manually updated the Pico with version 1.18. Surprisingly it is only the main PC. I tried another PC that didn't have Python on it and it worked fine. If I don't get this resolved I will go back to the Arduino and tell people to avoid the Pico altogether. 
[image]

owardlaw, 2022-04-23:
Ran into this issue on my windows 11 machine but works fine on my Macbook pro.

xXDerpyXx, 2023-01-07:
[image]
still happening to this day, windows 10, otherwise up to date

xXDerpyXx, 2023-01-07:
https://github.com/thonny/thonny/issues/1986

solution here ^^^^

this needs to be fixed by default, isn't python supposed to be a beginner language? and its not a hard fix either, you just need to update ssl certificates, which can be done with a single file in 2-3 seconds.

proximacentaury, 2024-03-29:
Can you explain the 2-3second process a newbie, who does not even heard about this problem? I have the sll certificate problem right now.