thonny/thonny #1623: module datetime not available

John2108, 2021-02-18:
import datetime as dt

current_datetime = dt.datetime.now()

print(current_datetime)

aivarannamaa, 2021-02-18:
Are you using Python 3 back-end or MicroPython? This module is not meant to be available in MicroPython.

John2108, 2021-02-18:
I am using micro python on a raspberry pi pico. and running Thonny on a raspberry pi 4b 8gb.
first time of using pico
best regards.
John

aivarannamaa, 2021-02-18:
You could add this module to Pico yourself (either by copying https://github.com/micropython/micropython-lib/blob/master/datetime/datetime.py as `/lib/datetime.py` or by installing `micropython-datetime` via Thonny's package manager (Tools => Manage packages).

It's not so rosy, though. One potential problem is that the module is huge by MicroPython standars. A more direct problem is that it doesn't work:

```python
>>> import datetime as dt
current_datetime = dt.datetime.now()
print(current_datetime)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/datetime.py", line 1919, in <module>
  File "/lib/datetime.py", line 1836, in _create
AttributeError: type object 'tzinfo' has no attribute '__new__'
```

So, it seems you need to either fix the module yourself or find another solution. I recommend https://forum.micropython.org/ for further help.

closed 2021-02-18 by aivarannamaa

John2108, 2021-02-18:
Hi Aivar thankyou so much, changed to using Python 3 back-end all is now good
Best regards
John