thonny/thonny #1624: Unable to create bytecode-cache directory, only when running script from IDE

thouseweart, 2021-02-19:
My apologies if this is just some quirk of Python that I don't understand.

I am trying to make a local cache of compiled .pyc files to speed up import statements. A new feature in Python 3.8 allows you to do this by setting sys.pycache_prefix before you import. 

Here is a short script called pycache_prefix_test.py which does this:
`from appdirs import AppDirs #cross-platform user-specific app data directories
cache_dir = AppDirs("python_bytecode","").user_cache_dir #cross-platform user-specific app data directories
import sys
sys.pycache_prefix = cache_dir #store compiled bytecode locally for each user, when running python on shared network drive. Should massively speed up imports after the first time the user runs the program.
print(f"sys.pycache_prefix: {sys.pycache_prefix}")

print("Importing libraries...")
import pandas #some large library
print(f"Finished importing libraries")`

If I run this script from the command line using:
`python pycache_prefix_test.py`
or
`ipython pycache_prefix_test.py`
everything works well, with the cache directory tree being created and populated.

But if I delete the cache directory tree and try to run this script from Thonny (using the same python install), the Thonny shell prints:
`%Run pycache_prefix_test.py`
and the cache directory tree is not created.

Thanks for your time!

thouseweart, 2021-02-19:
I doubt that this is related, but I'll mention it just in case... if I do:
    `for thing in sys.path: print(thing)`
in Thonny vs. command line python, all the entries are the same, but some of the paths printing in the Thonny shell have been (mostly) case-folded....

thouseweart, 2021-02-19:
I am able to successfully create the cache directory tree from Thonny if I 

`Run > Run current script in terminal  (Ctrl+T)`

so I'm pretty sure it has something to do with the Thonny Shell. 

My second comment about case-folded paths seems to be unrelated, as the paths printed with the (Ctrl+T) method are also case-folded.

thouseweart, 2021-02-19:
Running the script after opening:

`Tools > Open system shell...`

also worked successfully.

thouseweart, 2021-02-19:
Changing the python interpreter using:

`Run > Select interpreter... > Alternative Python 3 interpreter`

to a python interpreter separate from the one that runs Thonny did not solve the issue.

thouseweart, 2021-02-19:
Starting ipython from the command line and running:

`%run pycache_prefix_test.py`

works successfully.

aivarannamaa, 2021-02-19:
Thank you for the report! Also, thank you for educating me -- I didn't know about  `pycache_prefix`.

I could reproduce this, but at the moment I don't know what may cause it. I suspected Thonny's custom import handler, but the problem persisted even if I turned it off.

Then I suspected `exec` (Thonny always first creates a Python process with its own back-end machinery and then `exec`-s user code). It turned out, that if I `exec` your example code as string, then in Windows the cache directory doesn't appear even if I do it outside of Thonny. In Linux, on the other hand, even this works properly.

I'll give it some more thought later.

Can you tell me your final goal -- why do you want custom location for compiled files?

thouseweart, 2021-02-19:
Sure, no problem! Thanks for the formatting help btw, I didn't know how to get Github to color-code python.

It looks like you have narrowed down the issue! I was not able to reproduce this behavior though:

>  if I exec your example code as string, then in Windows the cache directory doesn't appear even if I do it outside of Thonny.

In Windows, using a miniconda install of python 3.8.5, starting python from the command line, and running:

```python
code = """ #pasted code here """
exec(code)
```

The cache directory was created for me, whereas running the same command from the Thonny shell (same interpreter) does not create the cache directory.

My motivation is to have a shared python install located on a shared network drive. I'd like to host several quick-and-dirty data processing scripts on the shared drive so that I can email non-technical users a link to a .bat script that "just works", without requiring installation or anything on their end, and without requiring packaging/freezing/compilation on my end. That way if I need to make a small tweak to one of the scripts, the change will be immediate for all users without requiring downloads, updates, etc.

The network latency seems to cause very long import times. For one of my scripts, the imports were taking ~50 seconds. If I cache the bytecode locally on my machine, the imports take ~5 seconds. I figure if I store a local cache of the files on each user's machine, running the script will be much faster after the first run.

thouseweart, 2021-02-19:
I tried unsuccessfully to compile to AST as an intermediate step (I'm far outside my area of expertise here, but it seemed like it was worth a shot). Here are several variations that I tried running from Thonny shell:

```python
code = """from appdirs import AppDirs #cross-platform user-specific app data directories
cache_dir = AppDirs("python_bytecode","").user_cache_dir #cross-platform user-specific app data directories
import sys
sys.pycache_prefix = cache_dir #store compiled bytecode locally for each user, when running python on shared network drive. Should massively speed up imports after the first time the user runs the program.
print(f"sys.pycache_prefix: {sys.pycache_prefix}")
print("Importing libraries...")
import pandas #some large library
print(f"Finished importing libraries")"""

import ast
tree = ast.parse(code)

#exec(compile(tree, filename="<ast>", mode="exec")) #didn't work

#exec(compile(tree, filename="<ast>", mode="exec", dont_inherit=False)) #didn't work

#flags = ast.PyCF_ALLOW_TOP_LEVEL_AWAIT | ast.PyCF_TYPE_COMMENTS
#exec(compile(tree, filename="<ast>", mode="exec", dont_inherit=False, flags=flags)) #didn't work
```

thouseweart, 2021-02-19:
I tried making the cache directory tree using cmd python, then deleting all the .pyc files, but leaving the directory structure intact. Then ran exec(code) from Thonny shell, but still no .pyc files were created.

thouseweart, 2021-02-19:
Interestingly, in the Thonny shell, after exec(code), if you do:

```python
import importlib
importlib.util.cache_from_source(pandas.__file__)
```

the path to the .pyc bytecode is the desired path (apart from the weird case-folding issue).

aivarannamaa, 2021-02-20:
I'll keep this issue open for later investigation, but it may come out that this behavior is caused by something in Thonny's back-end I'm not willing to give up.

In the meanwhile, I recommend you just instruct your users to run code with Ctrl+T

thouseweart, 2021-02-22:
Yep, makes sense to me, thanks for your help! I'll update here if I run more tests or discover anything relevant.