thonny/thonny #1625: Prevent involuntary changes in Python libraries

r0the, 2021-02-19:
Recently, a student accidentally destroyed her Thonny installation and had to reinstall Thonny. Here's how it works:

1. Perform a user-only installation on Windows -> All Thonny-installed files are writeable for user.
2. Run a program like
    ``` python
    from turtle import *
    forward("a bit")
    ```
3. Klick on the bottom-most link in the shell to the origin of the error (somewhere in turtle.py). -> Now System library turtle.py is open and editable in the editor.
4. Later, paste code in the editor tab of turtle.py and save.

I'm not sure what a good approach to prevent such a problem would be. Actually I see two possible solutions:

- Don't provide clickable links to library files in a stack trace.
- Don't allow to edit library files in Thonny.

aivarannamaa, 2021-02-19:
Thank you for the report!

I like the second suggestion more, but I wouldn't make it impossible to change library code, because occasionally I've instructed Thonny users to add a small fix to Thonny's code (which is in library location).

Also, I've seen that silently disabling something can create confusion for users. For example, I've received countless bug reports about debug buttons grayed out with MicroPython back-end, so I've decided to make them always enabled and present a notice about debug not being supported instead.

What do you think of presenting a confirmation dialog with strong warning when the user tries to edit a library file?

r0the, 2021-02-21:
Thanks for your reply. A warning before saving a library file seems a good idea to me.