thonny/thonny #1627: Object evaluation at MicroPython REPL gives too short REPL

aivarannamaa, 2021-02-19:
I was to eager in #1582

closed 2021-02-19 by aivarannamaa

Ivoz, 2021-02-20:
Hopefully this fixes for me, I am having to do `[print(x, end=', ') for x in dir(object)]` all over the place instead of `dir(object)` atm

aivarannamaa, 2021-02-20:
Funny, a `dir(..)` was how I discovered this problem :)

Regarding your float example -- I don't think this is related to this issue. I assume you're referring to the strange result in float => int conversion? Is it MicroPython? I know that most boards have only single precision floats or do float calculations in software. I recommend testing it outside of Thonny (eg. via Putty, minicom or tio)

Ivoz, 2021-02-21:
Ya I'm suspecting the second issue might be float precision

aivarannamaa, 2021-02-23:
@Ivoz , 3.3.5 is now out: https://github.com/thonny/thonny/releases/tag/v3.3.5

Please check whether you agree that the problem is solved.

Ivoz, 2021-02-26:
```py
>>> dir(Pin)
['__class__', '__name__', 'value', '__bases__', '__dict__', 'IN', 'IRQ_FALLING', 'IRQ_RISING', 'OPEN_DRAIN', 'OUT', 'PULL_DOWN', 'PULL_HOLD', 'PULL_UP', 'WAKE_HIGH', 'WAKE_LOW', 'init', 'irq', 'off', 'on']
```

This looks better