thonny/thonny #1629: Backend terminated or disconnected. Use 'Stop/Restart' to restart. - minimal example

amogorkon, 2021-02-19:
Hi, I finally figured out the issue Thonny is having with my code. Since I couldn't use Thonny's debugging facility I basically had to bisect all methods of the class in question and then added prints everywhere, that was fun ;-)

This is the code in question:

```
class Domain:    
    def __init__(self, name, low, high, *, res=1, sets:dict=None):
        self._name = name
        self._high = high
        self._low = low
        self._res = res
        self._sets = {} if sets is None else sets  # Name: Set(Function())

    
    def __repr__(self):
        return f"Domain('{self._name}', {self._low}, {self._high}, res={self._res}, sets={self._sets})"
    
    def __getattr__(self, name):
        if name in self._sets:
            return self._sets[name]
        else:
            raise AttributeError(f"{name} is not a set or attribute")

temp = Domain("Temperature", -80, 80)
```

It seems that while in step-by-step debugging, Thonny is trying to access __repr__ and __getattr__ and then is stuck in a loop of __getattr__ somehow, which crashes the backend - it never gets to actually call __init__.

aivarannamaa, 2021-02-20:
Thank you for the snippet!

I tried small-stepping through it (with F7) both with nicer and faster debugger, but met no obstacles. Which Thonny version and OS are you using? Which Thonny views do you have open?

amogorkon, 2021-02-20:
Hey,

I made a small screencast for you: https://www.dropbox.com/s/0bg2g3d18ngcra7/Thonny%20%20-%20%20C__Users_micro_Desktop_experiment.py%20%20%40%20%2021%20_%2038%202021-02-20%2013-37-07.mp4?dl=0

The specs according "About Thonny" are: Thonny 3.3.3 on Win10 64bit, Python 3.7.9 (32-bit) Tk 8.6.9

amogorkon, 2021-02-20:
I just installed Thonny 3.3.4 - the problem seems to be fixed in that version. Thanks :-)

aivarannamaa, 2021-02-20:
I'm afraid it isn't actually fixed in 3.3.4. You probably had also changed something else. 

I was able to reproduce it and it looks like Thonny's debugger calls `repr` on your `Doman` object before its constructor has finished, therefore `self._sets` on the first line of `__getattr__` causes recursive call to `__getattr__`.

amogorkon, 2021-02-20:
Strange. I can't recall having changed the code when I upgraded, but I was guessing something like was happening, yep.

aivarannamaa, 2021-02-20:
At the moment I don't have good ideas how to fix it in Thonny, as Thonny's nicer debugger is supposed to create a snapshot of the stack (including repr-s) on each possible breakpoint. I know, this really inefficient, but that's the price of having the possibility for stepping back during debugging. But even if I ditch "Step back", the user is still allowed to set a breakpoint in the constructor and expects to see repr-s of all locals, including `self`.

I found following warning in [`__getattribute__`-s documentation](https://docs.python.org/3/reference/datamodel.html#object.__getattribute__):

> In order to avoid infinite recursion in this method, its implementation should always call the base class method with the same name to access any attributes it needs, for example, object.__getattribute__(self, name).

I think this is relevant here as well, so I suggest you rewrite your `__getattr__` like this:

```
    def __getattr__(self, name):
        _sets = object.__getattribute__(self, "_sets")
            
        if name in _sets:
            return _sets[name]
        else:
            raise AttributeError(f"{name} is not a set or attribute")
```