thonny/thonny #1631: AttributeError: 'NoneType' object has no attribute 'lineno'

amogorkon, 2021-02-20:
Next one, this time with Thonny  3.3.4 ;-)

>>> %Debug .test.py
ERROR   thonny.backend: Exception in _trace
Traceback (most recent call last):
  File "C:\Users\micro\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1789, in _trace
    return self._trace_and_catch(frame, event, arg)
  File "C:\Users\micro\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1834, in _trace_and_catch
    self._handle_progress_event(frame.f_back, event, marker_function_args, node)
  File "C:\Users\micro\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1892, in _handle_progress_event
    self._save_current_state(frame, event, args, node)
  File "C:\Users\micro\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1942, in _save_current_state
    or not range_contains_smaller_or_equal(prev_root_expression, focus)
  File "C:\Users\micro\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 139, in range_contains_smaller_or_equal
    return range_contains_smaller(one, other) or one == other
  File "C:\Users\micro\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 123, in range_contains_smaller
    this_start = (one.lineno, one.col_offset)
AttributeError: 'NoneType' object has no attribute 'lineno'

aivarannamaa, 2021-02-20:
Here I also need the program you're debugging. Also, please tell me at what location this happens.

amogorkon, 2021-02-20:
I'm working on https://github.com/amogorkon/fuzzylogic.

The code I'm running is 

```
import classes, functions

temp = classes.Domain("Temperature", -80, 80)
hum = classes.Domain("Humidity", 0, 100)
motor = classes.Domain("Speed", 0, 2000)

temp.hot = functions.R(15,30)
temp.cold = functions.S(0,20)

hum.dry = functions.S(20,50)
hum.wet = functions.R(40,70)

motor.fast = functions.R(1000,1500)
motor.slow = ~motor.fast

Rules = classes.Rule({(temp.hot, hum.dry): motor.fast,
                      (temp.cold, hum.dry): motor.slow,
                      (temp.hot, hum.wet): motor.fast,
                      (temp.cold, hum.wet): motor.slow,
                      })

print(Rules({hum:50, temp:0}))
```

and it fails on the last line while step-by-step debugging, although it seems to recover and still print the correct value. 
If you want to reproduce it, here are the two files from the repo:

https://github.com/amogorkon/fuzzylogic/blob/master/src/fuzzylogic/classes.py
https://github.com/amogorkon/fuzzylogic/blob/master/src/fuzzylogic/functions.py

aivarannamaa, 2021-02-20:
Thanks!

It's the generator in the first statement of `Rule.__call__`.

Here is a smaller program, which demonstrates the same problem:

```
words = {"abc": 1, "cde": 2}
max(len(c) for c in words.keys())
```

I commited a partial fix, which should solve your case, but it's not a proper solution (generators are hard).

BTW. Do you know that Thonny also has a faster debugger (Shift+F5)? Your programs are pretty large, so I'm surprised that you want to bother with the nicer debugger. As I told before, it is very inefficient (and without regrets, because there is alternative, faster debugger).

solsword, 2021-03-08:
Just bumping to say that I (or more accurately, one of my students) ran into this bug and I can confirm that the fix listed in the patch above prevents the crash. I'll have students use the faster debugger if they run into this as a workaround, which did work without the patch, although because they're intro students they can really benefit from the features of the nicer debugger (they aren't actually writing generator code, but one of my libraries for the class has some in it).

Thanks for the quick work on this :)

aivarannamaa, 2021-03-08:
@solsword, which Thonny version are your students using? This issue is still open because I want to dig deeper here, but the quick fix was released in 3.3.5 (ie. is also present in the latest version)

solsword, 2021-03-08:
Looks like our links need upgrading; our course page links to 3.2.7, which is what I'm running right now; I can't remember if I updated that link this semester or last semester but I suspect it was last semester. Is there an easy upgrade path besides uninstall/reinstall?

Thanks for letting me know that the fix is in the latest version!

aivarannamaa, 2021-03-08:
> Is there an easy upgrade path besides uninstall/reinstall?

@solsword, uninstall is not required -- the installer takes care of removing old version.