thonny/thonny #1634: ERROR: Could not find a version that satisfies the requirement tensorflow (from versions: none) ERROR: No matching distribution found for tensorflow Process returned with code 1

ahmet-soner, 2021-02-20:


ahmet-soner, 2021-02-20:
Bu modüller yüklenemiyormu

aivarannamaa, 2021-02-20:
Assuming you are currently using 32-bit Python (the one bundled with Python is 32-bit). The solution is to install a separate 64-bit Python, make Thonny use it (Run => Select interpreter) and then install tensorflow. See also #862.

Please reopen, if your problem was about something else.

closed 2021-02-20 by aivarannamaa

ahmet-soner, 2021-02-20:
Aivar Annamaa
I am grateful to you
I wish I could be as knowledgeable as you
I've been trying to run chatterbot for 3 weeks and haven't found a correct
solution from anyone
A program like Thonny you gave us free from you and support you for Turkey
continued to extend my eternal gratitude and thanks
If you come to Turkey, please call me, I want you to buy a nice meal in
Istanbul

Aivar Annamaa <notifications@github.com>, 20 Şub 2021 Cmt, 22:55 tarihinde
şunu yazdı:

ahmet-soner, 2021-02-20:
Bu arada berbat ingilizcem için özür dilerim :)
Google translate kullanıyorum
Gerçekten İstanbul'a gelirsen gezdirebilirim
Keras, Tensorflowu yüklemeyi başardım, sorun 32 bit python olmasıymış

Ahmet Soner <prgfiles@gmail.com>, 21 Şub 2021 Paz, 00:39 tarihinde şunu
yazdı:

ahmet-soner, 2021-02-20:
By the way, sorry for my lousy english :)
I'm using google translate
If you really come to Istanbul I can take you around
Keras, I managed to install Tensorflow, problem was 32 bit python

Ahmet Soner <prgfiles@gmail.com>, 21 Şub 2021 Paz, 00:42 tarihinde şunu
yazdı:

aivarannamaa, 2021-02-21:
Thank you for the kind words!