thonny/thonny #1637: Thonny breaks remote WebREPL connections

seganku, 2021-02-20:
Thonny breaks remote WebREPL connections, and remote WebREPL connections break Thonny.  It seems ideal that these should be able to coexist somewhat .. maybe auto-off the raw-paste-mode when a remote connection attaches?

If Thonny is connected (via Socket), and a WebREPL session connects, that WebREPL session will be non-functional.  Additionally, Thonny's REPL session will become non-functional, crashing with any input.

Thonny 3.3.4 and MicroPython 1.14.
```
ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in _handle_normal_command
    response = handler(cmd)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 636, in _cmd_get_globals
    "{name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')}"
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 549, in _evaluate
    out, err = self._execute(script, capture_output=True)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 505, in _execute
    self._execute_with_consumer(script, consume_output)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 708, in _execute_with_consumer
    self._submit_code(script)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 567, in _submit_code
    self._submit_code_via_raw_paste_mode(to_be_sent)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 657, in _submit_code_via_raw_paste_mode
    self._raw_paste_write(script_bytes)
  File "C:\Users\montg\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 691, in _raw_paste_write
    raise AssertionError("Unexpected read during raw paste: {}".format(data))
AssertionError: Unexpected read during raw paste: bytearray(b'p')
term: Exception in write() method, deactivating: OSError: [Errno 104] ECONNRESET
�
```

It's obviously not a huge deal, and the easy work around is to not try to type in two places at the same time.

Thank you so much for the awesome product you've created here.  It's really something incredibly special!
-monte

IhorNehrutsa, 2021-02-20:
Also:
When you press the "Connect" button in WebREPL you get:
```
Password:                                                                       
WebREPL connected                                                               
>>>                    
```
If you try to connect WebREPL from another browser you get a message "Disconnected" and:
```
Concurrent WebREPL connection from ('192.168.43.72', 55625) rejected   
```
in the first browser.

See micropython\extmod\webrepl\webrepl.py

aivarannamaa, 2021-02-20:
I assume what @seganku meant, was concurrent serial and WebREPL connections.

At the moment I don't know how to detect existing previous connection from either serial or WebREPL connection.

@seganku, are you trying to use serial and WebREPL concurrently out of curiosity or for another purpose?

seganku, 2021-02-20:
I had mostly been using Thonny to edit/save/copy files to/from the MicroPython device, but most REPL was being done more by WebREPL.  It's easy to just click a bookmark without considering if that device happens to be currently plugged into the computer or not.  My primary issue is that it is annoying to have to reboot the device any time Thonny touches it, since Thonny breaks the WebREPL and doesn't leave it in a usable state.

Putty does not do this.

Would it be possible/feasible/reasonable to add an option to disable to RAW mode REPL?  Presumably it would be slightly slower, but it might be a worthwhile tradeoff in some instances.

aivarannamaa, 2021-02-21:
> Would it be possible/feasible/reasonable to add an option to disable to RAW mode REPL? Presumably it would be slightly slower, but it might be a worthwhile tradeoff in some instances.

Actually, there might be a way. I don't know how I forgot about it before. See https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration and try `submit_mode = paste`.

seganku, 2021-02-21:
I switched over to Linux, and I'm seeing another oddity.  I thought it was what @IhorNehrutsa was mentioning above, but after re-reading their comment, it's clear they were talking about two separate webpage connections.

Thonny 3.3.4 on Linux appears to auto-connect to WebREPL, even if Tools->Options->Interpreter is set explicitly to Serial.  If I disable `import webrepl; webrepl.start()`, Thonny still works.  If webrepl is enabled, any remote connections will be denied because Thonny has already grabbed it.  I am guessing that Thonny does this so devs don't have to repeatedly explain how users can add themselves to the `dialout` group?

seganku, 2021-02-21:
`submit_mode = paste` appears to work perfectly.

Thanks!

aivarannamaa, 2021-02-22:
@seganku, no, Thonny is not supposed to open a WebREPL connection when it is configured to use serial. If it happens like this, then it is a bug. I'll check later.

aivarannamaa, 2023-06-02:
I could not reproduce Thonny opening a WebREPL when it's configured to use serial.

closed 2023-06-02 by seganku