thonny/thonny #1640: Chatterbot

ahmet-soner, 2021-02-20:
Traceback (most recent call last):
  File "C:\Users\MYSTIC\Desktop\chatterbot\ChatBot.py", line 1, in <module>
    from chatterbot import ChatBot
  File "C:\Users\MYSTIC\Desktop\chatterbot\chatterbot.py", line 2, in <module>
    from chatterbot.storage import StorageAdapter
ModuleNotFoundError: No module named 'chatterbot.storage'; 'chatterbot' is not a package
**************************************************************************************************
I know it's a problem that doesn't concern Thonny
But I installed the chatterbot
I created the file named ChatBot.py in C: \ Users \ MYSTIC \ Desktop \ chatterbot folder, when I run it I get StrogeAdapter problem
I can't figure it out, you are knowledgeable people, please help me on this matter too.

aivarannamaa, 2021-02-21:
Actually, I prefer not to discuss general programming problems here. I suggest you consult a Python tutorial about modules and packages and/or ask your question in a general Python forum (or StackOverflow). A tip: when you are asking something, then always provide the code you are having problems with.

closed 2021-02-21 by aivarannamaa

ahmet-soner, 2021-02-21:
Thank you king
Just say if you want to look at my problem with teamviewer :)
You are loved Tohonny

Aivar Annamaa <notifications@github.com>, 21 Şub 2021 Paz, 08:58 tarihinde
şunu yazdı: