thonny/thonny #1643: Coronabot

ahmet-soner, 2021-02-21:
[image]

ahmet-soner, 2021-02-22:
I found the solution to the problem
************************************
Or try changing ...
time_func = time.process_time ()
insted of .. time_func = time.clock in
C: \ Users \ Admin \ AppData \ Local \ Programs \ Python \ Python38 \ Lib
site-packages \ sqlalchemy \ util \ compat.py

aivarannamaa, 2021-02-22:
I'm sorry, but I'm not here to help with general Python programming questions. Please reopen the issue if your code works properly outside of Thonny but fails in Thonny.

closed 2021-02-22 by aivarannamaa

ahmet-soner, 2021-02-22:
Okey

Aivar Annamaa <notifications@github.com>, 22 Şub 2021 Pzt, 08:40 tarihinde
şunu yazdı: