thonny/thonny #1645: cannot import name 'InterruptCommand' from 'thonny.common'

Xaxper, 2021-02-22:
Hello,

I'm trying to hook up my esp32 device with thonny on ubuntu. I get the error:

"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 1, in <module>
    from thonny.common import (
ImportError: cannot import name 'InterruptCommand' from 'thonny.common' (/usr/local/lib/python3.8/dist-packages/thonny/common.py)

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
"

I get this after connecting the MicroPython Firmware.
Ive already had a look at the file 'thonny.common' in the address specified, and 'InterruptCommand' is there already. Anyone know what I can do?

aivarannamaa, 2021-02-22:
You may have an older Thonny version somewhere, which is somehow messing things up.

If you can, then open Plug-ins manager (Tools => Plug-ins), and find `thonny` from the list. If it is installed under your home directory, then you should uninstall it.

If this is not possible, then close Thonny and execute `pip3 list --user` in Terminal. If `thonny` shows up, then do `pip3 uninstall thonny`.

If there is no `thonny` under user site-packages, then tell me your OS, Thonny version, and how you installed it.

Xaxper, 2021-02-22:
Hello, 
I got the following when i entered `pip3 list --user`

`Package   Version
--------- -------
bitstring 3.1.7  
ecdsa     0.16.1 
esptool   3.0    
reedsolo  1.5.4  
`

Im using Ubuntu Ubuntu 20.04.2 LTS, and using Thonny 3.2.7. Ive installed it using `sudo pip3 install thonny`

Xaxper, 2021-02-23:
Hello,

I have found another version of thonny installed on my device, but i do not know how to remove it. I have used the command `sudo pip3 uninstall thonny`, and got the successful uninstall message for the 3.3.4 version. After doing this, I checked and the app is still present on my pc. I retyped the command, and got a message `Found existing installation: thonny 3.2.7`. I would like to uninstall it, but I don't know how. Here's what I got from the terminal (note: I changed the device name to `"mypcname"` because Im posting it on the forum).

"mypcname":~$ sudo pip3 uninstall thonny
[sudo] password for kacper: 
Found existing installation: thonny 3.3.4
Uninstalling thonny-3.3.4:
  Would remove:
    /usr/local/bin/thonny
    /usr/local/lib/python3.8/dist-packages/thonny-3.3.4.dist-info/*
    /usr/local/lib/python3.8/dist-packages/thonny/*
  Would not remove (outside of prefix):
    /usr/bin/thonny
Proceed (y/n)? y
  Successfully uninstalled thonny-3.3.4
"mypcname":~$ sudo pip3 uninstall thonny
Found existing installation: thonny 3.2.7
Not uninstalling thonny at /usr/lib/python3/dist-packages, outside environment /usr
Can't uninstall 'thonny'. No files were found to uninstall.

aivarannamaa, 2021-02-23:
Most likely the 3.2.7 is installed by `apt`. You can remove it by `sudo apt remove thonny`.

After this I recommend you install Thonny by following terminal command:

```
bash <(wget -O - https://thonny.org/installer-for-linux)
```

This will give you a private Python installation for Thonny. You can still run your own programs with system Thonny if you wish (Run => Select interpreter).

aivarannamaa, 2021-02-26:
Were you able to solve this problem?

closed 2021-03-06 by aivarannamaa

h3retek, 2021-08-17:
> Were you able to solve this problem?

Yes, thank you