thonny/thonny #1646: line number field not wide enough when printing .py file with 1000 or more lines

ddelker, 2021-02-23:
When printing a Python program listing with 1000 or more lines, the line number field "spills over" the least-significant digit of the line number to the next line, which effectively double-spaces subsequent lines of the program. For example, a listing will look like this:
...
998  # lines 999 and below are fine
999  # but after this, things mess up
100
    0  # line 1000 goes here
100
    1  # line 1001 is here

I've tried changing Thonny's margins, but that has no effect on the line numbers. 

I could not find this addressed anywhere, but could someone help me with a solution or a work-around, please?  Thanks!!

closed 2021-02-24 by aivarannamaa

aivarannamaa, 2021-02-24:
Thank you for the report, @ddelker, and thank you for the solution, @gagzilla!

I added a small tweak from my side -- in order to save space with smaller programs, I made the width of line numbers column dependent on the total number of lines in the file.

The fix will be in Thonny 4.0.0, which won't be released too soon. @ddelker, if you need the solution sooner, then locate Thonny's code in your computer (Tools => Open Thonny program folder) and apply this small change in @gagzilla's PR by hand.

ddelker, 2021-02-24:
Thank you for your prompt response! 

 

I made the changes you indicated to the __init__.py and template.html files in the plugins/printing folders. Unfortunately, those changes kept the print function from working (CTRL-p did not open the print dialog). Also, the “min-width: 2em” parameter in my template.html file is “width: 2em”  By changing that value to “width: 3em” that corrected the long line number problem, so I am happy!

 

I thought you should know the results of my testing.

 

David

PR #1647 (2021-02-24): Fixing the print margin for numbers
The earlier version had a fixed margin of 2em. Switching to a minimum of 2em- this should scale to accommodate longer text for numbers beyond 1000. Fixes #1646