thonny/thonny #1648: problem when handling 'TextDelete'

Xrmolee, 2021-02-24:
i used Thonny on PI Pico, flow the code , i get crash .  
Thonny version: 3.3.4

PS.:code run in python  is correct .

def init_speed(start,end,level):
        rate = int(end -start)/level
        result = {}
        for i in range(1,level+2):
            result[i] = start + (i-1)*rate
        result[0]=0
        print(result)
        return result
    
if __name__=="__main__":
    my_speed = init_speed(10000,65535,5)
    print(my_speed)
    for i in range(10):
        print(my_speed)

Traceback (most recent call last):
  File "C:\Users\xrmol\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1708, in event_generate
    handler(event)
  File "C:\Users\xrmol\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 239, in text_deleted
    self.update_plotter()
  File "C:\Users\xrmol\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 253, in update_plotter
    self.plotter.update_plot()
  File "C:\Users\xrmol\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1792, in update_plot
    self.update_legend(data_lines, force_clean)
  File "C:\Users\xrmol\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1865, in update_legend
    fill=self.colors[i - 1 % len(self.colors)],
IndexError: list index out of range

aivarannamaa, 2021-02-24:
@Xrmolee , thank you for the report! I'll reopen it, because it is a problem that needs to be fixed.

closed 2021-02-24 by aivarannamaa

Xrmolee, 2021-02-25:
@aivarannamaa  
I have been upgrade to 3.3.5 , an other code  cause the same crash

import time
class my_ai:
    def __init__(self):
        self.motor1_speed = 0
        self.motor1_start = 10000
        self.motor1_end = 65535
        self.motor1_level = 50
        self.motor1_motor_is_running = False
        self.motor1_level_speed = self.init_speed(self.motor1_start,self.motor1_end,self.motor1_level)
          
    def init_speed(self,start,end,level):
        rate = int(end -start)/level
        result = {}
        for i in range(1,level+2):
            result[i] = start + (i-1)*rate
        result[0]=0
        print(result)
        return result
  
  

    def motor_status(self):
        i = 0
        while True:
            i = i + 1
            if i>10:
                break
            current_speed = self.motor1_level_speed[self.motor1_speed]
            print(self.motor1_level_speed,self.motor1_speed,current_speed)
            time.sleep(1)

mytest = my_ai()
mytest.motor_status()

aivarannamaa, 2021-02-25:
@Xrmolee, the fix was implemented after realeasing 3.3.5. You'll get the fix in 4.0.

The problem was in Plotter. If you don't need it, then the work-around is to just close it. You may need to restart Thonny to make sure Plotter is turned off.

Xrmolee, 2021-02-25:
thank u reply, turned off Plotter, is no more erros.

aivarannamaa, 2021-03-07:
@Xrmolee , the fix is now published as Thonny 3.3.6: https://github.com/thonny/thonny/releases/tag/v3.3.6