thonny/thonny #1650: Thonny 3.3.5 coding 8859-1 shows misspelled words

speedy-10, 2021-02-24:
Hi, if I use:

```
# -*- coding: iso-8859-1 -*-
# Autor: mk, Datum: 2.5.08; spr, 24.2.21

# import random
from random import *
print('Würfeln\n')

n = int(input('Wieviele Würfe = '))

i = 1
while i <= n:
    w = randint(1,6)
    print(w)
    i = i+1
```
I got the wrong output, see picture. (With utf-8 in the first line it works.)
If iI'm right, with thonny 3.3.4 it worked with # -*- coding: iso-8859-1 -*-

[image]

aivarannamaa, 2021-02-24:
Thank you for the report!

Please post the problematic script as an attachment. From the screenshot I get the impression that the file is actually encoded as UTF-8, but according to the hint on the first line, Thonny reads it as `iso-8859-1`.

Have you saved this file with Thonny? If yes, then this is still Thonny's mistake, as Thonny should have saved it as `iso-8859-1`.

aivarannamaa, 2021-02-24:
One more thing -- what do you get, when you enter `echo $LANG` in your Terminal?

speedy-10, 2021-02-24:
Hi, I've saved the programm ans started it with F5.

[zufall1.txt](https://github.com/thonny/thonny/files/6037293/zufall1.txt)

`echo $LANG`
`de_DE.UTF-8`

Regards

PS: Why can't I drop a *.py file, which would be native here?

closed 2022-03-24 by aivarannamaa