thonny/thonny #1651: Unable to launch Thonny on MacOS

sidyule, 2021-02-25:
[image]
I am unable to launch Thonny using macOS Big Sur. I am running a 2018 15" MacBook Pro with Intel processor.

I have tried a number of versions of Thonny from the latest down to thonny-3.3.1_alt.pkg with the same results.

The following is from Console as I try to launch Thonny from Applications:

default	18:57:23.894438+0000	runningboardd	Launch request for app<application.org.thonny.Thonny.57922270.57930261(502)> is using uid 502 (divined from auid 502 euid 502)
default	18:57:23.894552+0000	runningboardd	Executing launch request for app<application.org.thonny.Thonny.57922270.57930261(502)> (LS launch org.thonny.Thonny)
default	18:57:23.894609+0000	runningboardd	Creating and launching job for: app<application.org.thonny.Thonny.57922270.57930261(502)>
default	18:57:23.894634+0000	runningboardd	_mutateContextIfNeeded called for org.thonny.Thonny
default	18:57:23.902939+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] This process will not be managed.
default	18:57:23.902966+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] HOME is <private>
default	18:57:23.903007+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] TMPDIR is <private>
default	18:57:23.903029+0000	runningboardd	Now tracking process: [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]
default	18:57:23.903093+0000	runningboardd	Using default underlying assertion for app: [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]
default	18:57:23.903145+0000	runningboardd	Calculated state for app<application.org.thonny.Thonny.57922270.57930261(502)>: running-suspended (role: None)
default	18:57:23.903192+0000	runningboardd	Acquiring assertion targeting [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] from originator [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] with description <RBSAssertionDescriptor| "RB Underlying Assertion" ID:179-179-36266 target:2147 attributes:[
	<RBSDomainAttribute| domain:"com.apple.underlying" name:"defaultUnderlyingAppAssertion" sourceEnvironment:"(null)">,
	<RBSAcquisitionCompletionAttribute| policy:AfterApplication>
	]>
default	18:57:23.903328+0000	runningboardd	Assertion 179-179-36266 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]) will be created as active
fault	18:57:23.903688+0000	runningboardd	RunningBoard launch requested for identifier org.thonny.Thonny launched process with different identifier org.python.python
default	18:57:23.904633+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] reported to RB as running
default	18:57:23.907988+0000	loginwindow	-[PersistentAppsSupport applicationReady:] | App: Thonny, ready, updating active tracking timer
default	18:57:23.906312+0000	runningboardd	Calculated state for app<application.org.thonny.Thonny.57922270.57930261(502)>: running-suspended (role: None)
default	18:57:23.908051+0000	loginwindow	-[ApplicationManager checkInAppContext:eventData:] | ApplicationManager: Checked in app : Thonny
default	18:57:23.906384+0000	runningboardd	Acquiring assertion targeting [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] from originator [daemon<com.apple.coreservices.launchservicesd>:113] with description <RBSAssertionDescriptor| "foregroundApp:2147" ID:179-113-36267 target:2147 attributes:[
	<RBSDomainAttribute| domain:"com.apple.launchservicesd" name:"RoleUserInteractiveNonFocal" sourceEnvironment:"(null)">
	]>
default	18:57:23.907655+0000	runningboardd	Assertion 179-113-36267 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]) will be created as active
default	18:57:23.909838+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring jetsam update because this process is not memory-managed
default	18:57:23.910629+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring suspend because this process is not lifecycle managed
default	18:57:23.911417+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Set darwin role to: None
default	18:57:23.911979+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring GPU update because this process is not GPU managed
default	18:57:23.912825+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Skipping AppNap state - not lifecycle managed
default	18:57:23.913324+0000	runningboardd	Finished acquiring assertion 179-179-36266 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147])
default	18:57:23.913538+0000	runningboardd	Successfully acquired underlying assertion for [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]
default	18:57:23.914444+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring jetsam update because this process is not memory-managed
default	18:57:23.914752+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring suspend because this process is not lifecycle managed
default	18:57:23.914220+0000	runningboardd	Calculated state for app<application.org.thonny.Thonny.57922270.57930261(502)>: running-active (role: UserInteractiveNonFocal)
default	18:57:23.915056+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Set darwin role to: UserInteractiveNonFocal
default	18:57:23.915312+0000	Finder	LAUNCH: 0x0-0x39d39d org.thonny.Thonny starting stopped process.
default	18:57:23.915487+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring GPU update because this process is not GPU managed
default	18:57:23.914201+0000	runningboardd	Acquiring assertion targeting [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] from originator [daemon<com.apple.coreservices.launchservicesd>:113] with description <RBSAssertionDescriptor| "foregroundApp:2147" ID:179-113-36268 target:2147 attributes:[
	<RBSDomainAttribute| domain:"com.apple.launchservicesd" name:"RoleUserInteractiveNonFocal" sourceEnvironment:"(null)">
	]>
default	18:57:23.916353+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Skipping AppNap state - not lifecycle managed
default	18:57:23.916717+0000	runningboardd	Assertion 179-113-36268 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]) will be created as active
default	18:57:23.918927+0000	runningboardd	Finished acquiring assertion 179-113-36267 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147])
default	18:57:23.919492+0000	runningboardd	Acquiring assertion targeting [app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] from originator [daemon<com.apple.coreservices.appleevents(55)>:345] with description <RBSAssertionDescriptor| "AE/Thonny" ID:179-345-36269 target:2147 attributes:[
	<RBSDomainAttribute| domain:"com.apple.launchservicesd" name:"LSNotification" sourceEnvironment:"(null)">
	]>
default	18:57:23.919632+0000	runningboardd	Assertion 179-345-36269 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]) will be created as active
default	18:57:23.920634+0000	runningboardd	Finished acquiring assertion 179-113-36268 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147])
default	18:57:23.921087+0000	runningboardd	Calculated state for app<application.org.thonny.Thonny.57922270.57930261(502)>: running-active (role: UserInteractiveNonFocal)
default	18:57:23.921066+0000	runningboardd	Invalidating assertion 179-113-36267 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147]) from originator [daemon<com.apple.coreservices.launchservicesd>:113]
default	18:57:23.921303+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring jetsam update because this process is not memory-managed
default	18:57:23.921251+0000	runningboardd	Finished acquiring assertion 179-345-36269 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147])
default	18:57:23.921674+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring suspend because this process is not lifecycle managed
default	18:57:23.922238+0000	runningboardd	[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147] Ignoring GPU update because this process is not GPU managed
[... cut ...]
default	18:57:23.997437+0000	loginwindow	-[PersistentAppsSupport applicationQuit:] | App: Thonny, quit, updating active tracking timer
error	18:57:24.044544+0000	runningboardd	RBSStateCapture remove item called for untracked item 179-113-36267 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147])
error	18:57:24.044615+0000	runningboardd	RBSStateCapture remove item called for untracked item 179-113-36268 (target:[app<application.org.thonny.Thonny.57922270.57930261(502)>:2147])

aivarannamaa, 2021-02-25:
Thank you for the report!

Have you been able to run Thonny with Big Sur on this computer before? If yes, then have you recently updated your OS? Have you tried restarting your computer?

Please try running Thonny from Terminal:

```
/Applications/Thonny.app/Contents/MacOS/thonny
```

Do you get any output?

Please go to `~/Library/Thonny`. Does this directory exist? Do you see any errors in any of the log files in this directory?

sidyule, 2021-02-25:
I haven't been able to run it on Big Sur at all. I have only just started trying to use it. I have restarted the computer.

Running it from the Terminal gave me some errors ending with:
```
PYTHONPATH is currently: "/usr/local/lib/python2.7/site-packages"
     You should `unset PYTHONPATH` to fix this.
```

I have run unset PYTHONPATH and can now run Thonny using 
`/Applications/Thonny.app/Contents/MacOS/thonny`

I can't run Thonny by opening it from the GUI though.

aivarannamaa, 2021-02-25:
This is good to know!

How did you unset `PYTHONPATH`? Only for the same Terminal session?

If you could find out where this variable gets set, then you might be able to remove it and start Thonny normally (possibly after a restart). Unfortunately I'm not too familiar with macOS, so I don't know where to look.

sidyule, 2021-02-25:
I just ran `unset PYTHONPATH` I had been installing the latest version of Python and setting up my environment to get started so not sure if that caused confusion. I haven't had chance to reboot to see if the change is persistent. What I'm not sure about is why I can run it from the cli but not from the GUI, although I am happy with that. I'm just getting started with using Thonny for coding against the Pico Pi.

aivarannamaa, 2021-02-25:
> I just ran `unset PYTHONPATH`

This affected only your Terminal session, ie. it won't have persistent effect neither for running Thonny from Terminal nor for launching it from Finder/Dock/Desktop.

I had been installing the latest version of Python and setting up my environment to get started so not sure if that caused confusion

Python 3, right? For some reason, the error mentioned Python 2.7. It may be some other app which has set this environment variable, but I don't know where.

You could look into `~/.bashrc` and `~/.bash_profile` to see if there is anything related to `PYTHONPATH` there.

aivarannamaa, 2021-02-26:
Are you using Homebrew in this machine? If yes then for which apps?

sidyule, 2021-02-26:
I have used homebrew to install a few bits as I was using VSCode and needed npm and some other bits. I am busy going through pulling that all out now as I don't need all of it. I ham also hunting for what is setting `PYTHONPATH`. I can't find it in any of the usual places but this looks like it is definitely my environment causing the issue. Really appreciate your help on this.

sidyule, 2021-02-27:
This is resolved now. I managed to get Thonny working fine but still had issues with PYTHONPATH being set so followed this guide https://support.enthought.com/hc/en-us/articles/204469160-How-do-I-set-PYTHONPATH-and-other-environment-variables-for-Canopy- and creating the environment.plist file. Thanks for helping guide me in the right direction.

aivarannamaa, 2021-02-27:
Thanks a lot -- this article was an interesting read.

I'd keep this issue still open for now, as I'm hoping I can add something to Thonny's launcher such that unsuitable environment variables become automatically hidden for Thonny. 

Did you find out where was your `PYTHONPATH` being set? Or can you tell me how to reproduce your problem?

sidyule, 2021-02-27:
I didn't quite get to the what was setting `PYTHONPATH` although wondered if it is the system default one. The only thing I have really done is install Python IDE from python.org, added a couple of extensions to VSCode (now removed), Pymakr and node.js via homebrew. I did try using pyenv when I first had the issue but this didn't seem to help.

I have been through a few uninstalls of various bits to replicate but can't seem to quite find any combination other than by using the suggestion in the link above.

aivarannamaa, 2021-03-02:
Note to myself: need to shadow selected `PYTHON*` variables in Thonny's Info.plist: https://developer.apple.com/library/archive/documentation/General/Reference/InfoPlistKeyReference/Articles/LaunchServicesKeys.html#//apple_ref/doc/uid/20001431-106825

Ivoz, 2021-03-07:
@sidyule you can look in files in your home directory for scripts like `~/.bashrc` and `~/.bash_profile`, possibly in the `~/.config` folder, or any other profile / rc / alias looking files (they can be hidden by default, this is what the `.` at the start of the name does; use `ls -al` in the terminal to see everything). These files are shell script files that are usually run when your system logs in, and can set environment variables like this.

One problem I see a lot of (at least historically) is after people have installed / uninstalled dev environments many times, often the installers might not remove everything they have set, so a spare environment variable that you really don't want any more gets left in one of these files.

aivarannamaa, 2021-03-07:
@sidyule, I released 3.3.6 which should hide PYTHONPATH from Thonny: https://github.com/thonny/thonny/releases/tag/v3.3.6

I'm leaving this issue still open, as I'm considering hiding more `PYTHON*` variables in 4.0

sidyule, 2021-03-07:
> @sidyule you can look in files in your home directory for scripts like `~/.bashrc` and `~/.bash_profile`, possibly in the `~/.config` folder, or any other profile / rc / alias looking files (they [...]

I have checked these files and can't find anything in them. I am using zsh and checked the dot files for that too and can't find anything. Pretty sure it is as you described with installing/uninstalling things but just can't get to the bottom of where it is coming.

sidyule, 2021-03-07:
> @sidyule, I released 3.3.6 which should hide PYTHONPATH from Thonny: https://github.com/thonny/thonny/releases/tag/v3.3.6
> 
> I'm leaving this issue still open, as I'm considering hiding more `PYTHON*` variables in 4.0

Thanks for this. I have been using Thonny with no issues but have updated and all is good still.

aivarannamaa, 2021-12-26:
#2038 subsumes this issue

closed 2021-12-26 by aivarannamaa