thonny/thonny #1653: Introduce "Rename" refactoring

aivarannamaa, 2021-02-26:


FranckCHAMBON, 2021-03-18:
I'd like to see the rename feature in the 'Variables' display area, at right.
A single right-click on the name open a menu 'rename variable', and go... Or a DoubleClick to enter rename field.

I think it's a wonderful feature for beginners.
Imagine he wrote a big code with a variable named `n` and the teacher ask him a much better name...
He can't use a replace_text("n", "better_name"), it would break the code...

In the same way, I'd like to be able to select (copy, ...) the Value field of a variable.

Thanks,

aivarannamaa, 2021-03-18:
I'm afraid renaming variable in the Variables tables would confuse matters a lot. Variables table is part of program's run-time face, but when we're renaming a variable, we want to change program's static face, ie. source code. 

It is still possible to do something smarter than text replacement. The goal of this issue is to implement variable rename at the source level such that when you rename a local variable `n`, then it really replaces the occurrences of this local variable, not a `n` in another name nor a global variable `n`.