thonny/thonny #1660: Thonny backend crashes when connecting to Raspberry Pi Pico

brianmcnett, 2021-02-27:
This may not be an issue with Thonny itself. It might just be my install. However, I've found no way to recover from it. Error message Below:
===

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - use Stop/Restart to interrupt more and enter REPL.

ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 136, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 155, in _prepare_after_soft_reboot
    self._execute_without_output(script)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 528, in _execute_without_output
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:

class __thonny_helper:
    try:
        import uos as os
    except ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = dict()
    last_repl_values = []
    @classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.last_repl_values.append(obj)
            cls.last_repl_values = cls.last_repl_values[-5:]
            print('[object_link_for_thonny]', cls.repr(obj), '@', id(obj), '[/object_link_for_thonny]', sep='')

    @staticmethod
    def print_mgmt_value(obj):
        print('<thonny>', repr(obj), '</thonny>', sep='', end='')

    @staticmethod
    def repr(obj):
        try:
            s = repr(obj)
            if len(s) > 1000:
                s = s[:1000] + "..."
            return s
        except Exception as e:
            return "<could not serialize: " + str(e) + ">"

    @classmethod
    def listdir(cls, x):
        if hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @classmethod
    def getcwd(cls):
        if hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2021-02-27:
Have you been able to use Pico from Thonny before? With the same computer? macOS, I presume, but which version?

Can you connect to the MicroPython REPL with `screen`? Instructions:

* Close Thonny
* reset Pico (unplug and plug it back)
* Open Terminal
* enter `screen /dev/cu.usbmodem*`
* Press ENTER

Do you have something in the `main.py` on your Pico? (If you don't know, then probably not).

brianmcnett, 2021-02-27:
I've not been able to use Pico from Thonny before at all. Following your instructions, produces the following output:

>>>                     return ""    @classmethod   ttr(cls,   r5D

I've tried reflashing the pico with no result. This would, however, appear to rule out Thonny as the source of my issue, correct?

aivarannamaa, 2021-02-27:
If you did reset your Pico after closing Thonny and before connecting via `screen`, then this output would be surprising.

If you forgot the reset, then it may be leftovers from the conversation with Thonny.

If you press ENTER several times, can you get into the state where you can properly evaluate some Python expressions?

brianmcnett, 2021-02-27:
After pressing return several times, MicroPython does a soft reboot, and I get this:

MPY: soft reboot
MicroPython v1.14 on 2021-02-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>
MicroPython v1.14 on 2021-02-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== 
Any further action results in what looks like snippets of MicroPython code, which, being incomplete, do not evaluate and eventually generate a syntax error. At this point, MicroPython does a soft reboot and enters paste mode, etc.

aivarannamaa, 2021-02-27:
I see. If you got this after un-plugging and re-plugging Pico (ie. by cleaning whatever Thonny may had left in Pico's RAM), then it's either your USB driver's problem or faulty hardware.

I recommend asking for advice in the RPi section of MicroPython forum: https://forum.micropython.org/viewforum.php?f=21

brianmcnett, 2021-02-27:
Thanks. I did a complete nuke of everything on the Pico and reinstalled microPython.

I'm happy to report that I can now execute code in REPL via screen. I don't anticipate any issues with Thonny. We can mark this as closed.

closed 2021-02-27 by brianmcnett

aivarannamaa, 2021-02-27:
Great, this is good to know! 

Please test with Thonny and re-open the issue if something doesn't look right!