thonny/thonny #1661: Modifier ⌥ on Macs

juanfal, 2021-02-28:
Hello, ⌥ key (as they are used for uncomment/comment in Thonny) can’t be used in OSX as they are simple letters, alternative, but never sent to menus and then written and catastrophically destroying the text you wanted to comment/uncomment

I’d suggest alternatives to ⌘ 3 (⌘ ⌥ 3 and ⌘ ⌥ ⇧ 3) for force commenting and uncommenting 

Another glitch with Macs is the disaster of Tcl/Tk crashing each time you do type

aivarannamaa, 2021-03-01:
Thank you for the report! 

I don't know where I took these shortcuts from. IDLE currently has ctrl-3 and ctr-4... Anyway, I'll deal with this.

> Another glitch with Macs is the disaster of Tcl/Tk crashing each time you do type

I assume you meant the problem described at https://core.tcl-lang.org/tk/tktview/1626ed65b879bd76db99ef6abbbe18b64bef95a5

aivarannamaa, 2021-03-06:
To be handled together with #1244

aivarannamaa, 2022-04-01:
I'll leave these as is for now, because different IDE-s doesn't seem to agree on which shortcut to use for adding / removing comments. In 4.1 I hope to implement the possibility to assign custom shortcuts (#306) and then I'll remove these default shortcuts altogether.

The crash you are referring to shouldn't occur anymore with newer Tk in Thonny 4.0

closed 2022-04-01 by aivarannamaa

juanfal, 2022-04-01:
Thanks so much!!!

I didn’t try the ⌘-[ crashing shortcut (so used to this disaster!).  First time, in years we won’t need to explain our students they have not to press that shortcut.

THANKS!!!!

PS.:  ⌘-3 for switching comment is more than enough, it works like a charm