thonny/thonny #1662: Unable to save file on Pico Pi using CircuitPython

bitboy85, 2021-02-28:
Maybe this is a stupid beginner issue, but i'm unable to save a script on the Pico Pi using Thonny.
Every time i try it gives me the following error

ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Error Could not open file /code.py for writing, output:
[Errno 30] read-only file system

I'm using Linuxmint 20 cinnamon and within nemo i can create and delete files.
Thonny version is 3.3.5 installed with pip3
CircuitPython: Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Raspberry Pi Pico with rp2040

Thonny is able to read files and execute code on the device, but unable to store.
When connected, the device mounts as usb storage device.

aivarannamaa, 2021-02-28:
Thank you for the report!

I assume when you opened the file, you chose the option for loading from "CircuitPython device" instead of "This computer". 

I'll try to reproduce your case, but in the meanwhile you can close this code.py, and load it from "This computer" -- just locate the drive that Pico creates and find code.py from there.

thlabbe, 2021-03-20:
Same problem here, ... if it could help :
```
D:\PiPico\project\PiPico_maquereau\rotary_tla.py (3067 bytes)
ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in _handle_normal_command
    response = handler(cmd)
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1026, in _cmd_upload
    return super(BareMetalMicroPythonBackend, self)._cmd_upload(cmd)
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 297, in _cmd_upload
    pathlib.PurePosixPath,
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 385, in _transfer_files_and_dirs
    transfer_file_fun(item["source_path"], item["target_path"], copy_bytes_notifier)
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 405, in _upload_file
    callback,
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1190, in _write_file
    self._write_file_via_serial(source_fp, target_path, file_size, callback)
  File "C:\Users\tlabbe\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1248, in _write_file_via_serial
    "Could not open file %s for writing, output:\n%s" % (target_path, out + err)
RuntimeError: Could not open file /rotary_tla.py for writing, output:
[Errno 30] Système de fichier en lecture seule
``` 
REPL : ok
Windows 10 (64 bits)
Python 3.7.9 (32 bits)
Thonny 3.3.6
Adafruit CircuitPython 6.2.0-beta.4 on 2021-03-18; Raspberry Pi Pico with rp2040

Workaround : copy/paste from this computer to the drive CIRCUITPY using Windows Explorer

aivarannamaa, 2021-03-21:
@thlabbe, I found the reason why Thonny failed in your case -- it looked for "read-only" in the error message, but I had forgotten that the error message could be in another language. I added a fix for this.

@bitboy85, was your error message originally in English or did you translate it for me?

bitboy85, 2021-03-21:
Its mixed.
[image]

aivarannamaa, 2021-03-21:
@bitboy85, Thanks! "Schreibgeschützte Dateisystem" explains this, ie. your case will be also fixed in the next release.

closed 2021-03-21 by aivarannamaa

callimero, 2021-04-12:
I just also stumbled on that error. It also don't work in 3.3.3 and 3.3.5 I wonder how that could not have been found?! Is everyone only edit the code.py in which case it works... Sorry, about the rant trying to write an article, now have to go with "Mu".

aivarannamaa, 2021-04-12:
@callimero, if you can edit code.py and not other files, then you may have a different issue. 

Do you have English variant of CircuitPython or something else?

callimero, 2021-04-26:
Same error as aivarannamaa above. The german part comes from my german windows 10.

It tries to save a "/File.py" is maybe the reason.  

If I just use a different Editor I can save fine to my Pico. AND then I can open it with Thonny and also SAVE it. "Magic", so the error ist only with new files saved first time in Thonny.

callimero, 2021-04-26:
Just to let you know, I am writing for a german magazine and need to get this working for our readers before issue is printed :)

callimero, 2021-04-26:
Are there pre build release candidates or daylies of Thonny?

aivarannamaa, 2021-04-27:
> The german part comes from my german windows 10.

I'm not sure about this. It is the case if you are sure you have English variant of CircuitPython. As a test you can try following piece of code on CircuitPython REPL:

```
k = open("dummy.txt", "w")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 30] Read-only filesystem
```

If you get German message, then you have German variant of CircuitPython. If not, then I'm really confused.

> Are there pre build release candidates or daylies of Thonny?

Unfortnately not.

As a work-around, you can save and open the files from the CircuitPython drive directly. For example, when you create a new untitled file in Thonny and save it then you should get a dialog, which asks you whether to save to local computer or CircuitPython device. Instead of choosing CircuitPython here, you could choose local computer and browse for mounted CircuitPython drive instead.

The background: by default CircuitPython is configured such that the files are writable via mounted drive and are read-only when writing them via CircuitPython code (like in the example snippet above). It is possible to reverse this (https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage). Most MicroPython devices don't have mounted drive at all so all writing must be done via MicroPython code. Thonny wants to support all cases, so it first tries to write the file via code and when it gets read-only error, it falls back to using mounted drive. The mistake was to assume the error is always in English.

callimero, 2021-04-30:
I patched my Thonny for now with the code from your commit. However I can't put that in a beginners article...

I just want to ask: will there a new release in the next 4 weeks? If not I am a bit in a trouble :/

**Ohh** Just realized what you told me with "As a work-around, you can save and open the files from the CircuitPython drive directly." Well that may be a valid workaround. However I am sure severeal readers will do it the other way anyhow :(

callimero, 2021-04-30:
Also decommented 

`    def _get_fs_mount_label(self):
        # This method is most likely required with CircuitPython,
        # so try its approach first
        # https://learn.adafruit.com/welcome-to-circuitpython/the-circuitpy-drive
        result = self._evaluate(
            dedent(
            try:
                from storage import getmount as __thonny_getmount
                try:
                    __thonny_result = __thonny_getmount("/").label
                finally:
                    del __thonny_getmount
            except ImportError:
                __thonny_result = None 
            except OSError:
                __thonny_result = None 
            
            __thonny_helper.print_mgmt_value(__thonny_result)
            
            del __thonny_result
            )
        )`

and that also helps I think.

aivarannamaa, 2021-04-30:
@callimero, alright, I'll make a release with selected bugfixes tonight or tomorrow.

aivarannamaa, 2021-04-30:
@callimero, ... regarding decommenting -- actually the code was not commented out. It is sent from Python process to CP device as string literal, that's why it was in quotes.

steeley, 2022-01-22:
Still get this error with Pico and CircuitPython.( I can use a normal text editor to read and write files on the device ok)

```
ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Error [Errno 1] Operation not permitted: '/Volumes/CIRCUITPY/greet.py'
```

Thonny3.3.13.  OSX Big Sur 11.6.2

RnoldR, 2024-02-18:
I am having this error too with Thonny version 4.1.4, see code below. It happens when I run from my local drive, from the rpico drive itself or use this code as the code.py code. Am I doing something wrong?

```
with open('logfile.txt', 'w') as outfile:
    outfile.write('Hi there\n')
```
Thonny Reponse:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 30] Read-only filesystem
```

aivarannamaa, 2024-02-25:
@RnoldR, your case is different. With CircuitPython you can either have its filesystem exposed as writable mounted volume or it can be writable via CircuitPython code, not both.