thonny/thonny #1665: What are the command line options for launching Thonny to go to specific line number

PySimpleGUI, 2021-02-28:
I've looked through the docs, github, stack overflow, etc, and I've not been able to find how to do this.

I would like to invoke thonny to open a file and go to a specific line number

For example, on PyCharm the format of the command to launch is:
pycharm  --line 1234 file.py

For Sublime it's
subliime file.py:1234

Can you tell me what the format is for Thonny?

aivarannamaa, 2021-02-28:
Thonny does not have such option, but it's good idea and I can add it in the next version.

PySimpleGUI, 2021-02-28:
Here's why I want it..... I'm integrated with PyCharm, Wing, Visual Studio, Sublime, and would like to add Thonny as well

[image]

PySimpleGUI, 2021-02-28:
If you can tell me the format of the command line, I'll add it to the list that is automatically filled in for each of the IDEs/editors.

I've figured out all of them for the "Top 11" Python editors and have them in this dictionary:
```python

    editor_format_dict = {
        'pycharm':'<editor> --line <line> <file>',
        'notepad++':'<editor> -n<line> <file>',
        'sublime':'<editor> <file>:<line>',
        'vim':'<editor> +<line> <file>',
        'wing':'<editor> <file>:<line>',
        'visual studio':'<editor> <file> /command "edit.goto <line>"',
        'atom':'<editor> <file>:<line>',
        'spyder':'<editor> <file>',
        'thonny':'<editor> <file>',
        'pydev':'<editor> <file>:<line>',
        'idle':'<editor> <file>'}
```

**I don't yet know: Spyder, Idle and Thonny**

aivarannamaa, 2021-02-28:
I like the colon variant most, so let's agree on this. 

Do you expect Thonny executable to be on the PATH or you're probing different locations?

PySimpleGUI, 2021-02-28:
It's up to the user to determine how the editor is invoked, but I'm assuming it'll be the fully path.  It doesn't have to be though.  Whatever is input into the "Editor Program" input will be used as the editor.

For example, I'm using PyCharm, so a fully qualified path to their batch file is needed.

[image]

For the tests I've run, I've been using the full path to the different editors/IDEs.

PySimpleGUI, 2021-03-16:
If you end up releasing something, please respond here so that I know to add it to the PySimpleGUI code.

aivarannamaa, 2021-03-16:
> If you end up releasing something, please respond here so that I know to add it to the PySimpleGUI code.

Will do

PySimpleGUI, 2021-03-16:
At the moment, the editing is limited to the file level.  Here's how it looks when using the PySimpleGUI Demo Browser.

[image]

PySimpleGUI, 2021-03-16:
Oh, I suppose I should have shown the use case where the line number is actually used.

It's in the error messages that line numbers come into play.

In this screenshot, I've setup the global editor to be Thonny.  This means it'll be used in error handling.  Since it's Thonny, I can't yet bring users to the line number, but I can bring them to the correct file.

[image]

aivarannamaa, 2021-03-16:
Thank you for this explanation!