thonny/thonny #1667: Problems to run on Windows 10

Abenene, 2021-02-28:
Hello,
I´ve a problem in my installation on Windows 10 for raspberry pi pico,
This is the message in the console:
Traceback (most recent call last):
  File "C:\Users\Josep\AppData\Local\Programs\Thonny\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Users\Josep\AppData\Local\Programs\Thonny\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Josep\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\__main__.py", line 40, in <module>
    thonny.set_dpi_aware()
  File "C:\Users\Josep\AppData\Local\Programs\Thonny\lib\site-packages\thonny\__init__.py", line 362, in set_dpi_aware
    import ctypes
  File "C:\Users\Josep\AppData\Local\Programs\Thonny\lib\ctypes\__init__.py", line 551, in <module>
    _reset_cache()
  File "C:\Users\Josep\AppData\Local\Programs\Thonny\lib\ctypes\__init__.py", line 273, in _reset_cache
    CFUNCTYPE(c_int)(lambda: None)
MemoryError

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
Many thanks in advance
Josep

aivarannamaa, 2021-03-03:
Thank you for the report!

Does it happen always? Have you been able to run Thonny properly on this computer before? 

Note to myself: https://bugs.python.org/issue35523

aivarannamaa, 2021-11-13:
Have you been able to use Thonny after your report?

closed 2021-11-17 by aivarannamaa